LaserWeb/LaserWeb4 #255: Per Axis Homing comms

openhardwarecoza, 2017-04-09:
https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/jog.js#L350 
https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/jog.js#L373
https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/jog.js#L395

Doesnt work yet (:

Havent checked if Probe works, but probably not, as I dont see any code for it yet (:

Not too big a rush

cprezzi, 2017-04-10:
Yep, I did not implement it, because at that time I only had grbl, which doesn't support single axes homing.
Will add it when I get some time.
Should I hide those menu entries until the feature is working?

openhardwarecoza, 2017-04-10:
No need, just logging issues as I find things (:

Leave 'em for now, when we do implement it we can hide those for Grbl

openhardwarecoza, 2017-06-01:
https://plus.google.com/111079322062040624414/posts/WN1uMd7pJMJ - another user needing this (please point Walter at the open issue - its not implemented yet)

openhardwarecoza, 2017-06-01:
@cojarbi  ^ (; please point Walter

cprezzi, 2017-06-12:
I have implemented the home() command on server side (firmware dependant). Now I need to add it to the frontend jog tab.

cprezzi, 2017-06-14:
@domenic-d We habe the settings 'Tool Offset' and 'Default Start Height' for the Z-Stage. Can you please explain me, how those params are used?
I'm implementing probing and am courious if I can reuse one of those params for the offset.

domenic-d, 2017-06-14:
Tool-Offset is used to compensate for slight differences with respect to the height of the laser. We have a calibration process that allows customers to determine their optimal laser focus. This value they determine is then used as an offset to the laser height (z) value.

Start height is basically the distance to the top of the material from the base (z=0) of the machine. This allows users to type the thickness of their material and the z-height is set accordingly.

domenic-d, 2017-06-14:
Z value = Start_Height + Tool_Offset

cprezzi, 2017-06-14:
@domenic-d Thank you for clarifying.

jorgerobles, 2017-07-10:
Tested. Closing :)

closed 2017-07-10 by jorgerobles