LaserWeb/LaserWeb4 #256: Live jogging should send jog command instead of gcode and respect feed units

cprezzi, 2017-04-10:
@jorgerobles Live jogging is directly sending G0 commands instead of using the jog function, which is using the correct method depending on firmware (on server side). Also feed units should be respected.

jorgerobles, 2017-04-10:
hmm Jog is using relative coordinates, and only works an axis... Feed is true that has to be honored. But  I don't see harm on G0. What firm does it different? (just to know, anyways.)

tbfleming, 2017-04-10:
grbl has a special jog command that's interruptible.

cprezzi, 2017-04-10:
Grbl has a special jog command that preserves parser state and can be cancelled correctly. It can be used with absolute or relative distance (G90/G91). It's also safer because jog is only accepted in idle or jog state.

cprezzi, 2017-04-10:
see: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

cprezzi, 2017-04-10:
G0 is fixed feed on Grbl, only Smoothie allows to set feed on G0.

jorgerobles, 2017-04-10:
ok, will use new jogTo Method on com_refactor :)

cprezzi, 2017-04-10:
The server decides depending on the firmware, which method has to be used. 

I probably need to extend the jog command on server side. At the moment it only supports one axes at a time and relative distance. ;) (see https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging)

jorgerobles, 2017-04-21:
@cprezzi would you please implement `jogTo` to use it instead current gcode?

cprezzi, 2017-04-21:
@jorgerobles Do you need absolute and/or realtive mode for distances?

jorgerobles, 2017-04-21:
@cprezzi, I currently use the XY Vector projection of @tbfleming workspace.
That are absolute coordinates of the render.  I suppose I can make diff
with current work position, to make them relative.

cprezzi, 2017-04-21:
@jorgerobles I can provide a jogTo() function with the following params: x, y, z, mode, feed.
Mode would be 0 for absolute and 1 for relative (corresponding to G90 and G91). Is that enough?

jorgerobles, 2017-04-21:
Perfect!!

cprezzi, 2017-04-21:
Do we also need A, B, and C axis for jogging?

jorgerobles, 2017-04-21:
hmm.. A maybe

jorgerobles, 2017-04-21:
I didn't mind on it O_o

cprezzi, 2017-04-21:
I leave abc out for the moment, as we also don't have jog buttons for abc and the workspace only knows xyz.

cprezzi, 2017-04-21:
Done with lw4/dev-es6 commit https://github.com/LaserWeb/LaserWeb4/commit/9d967de6f27c93b907c0d149cec5b554301fe0de and lw.comm-server/master commit https://github.com/LaserWeb/lw.comm-server/commit/e0282c2dcfa2b5dde89b1b1848ce22e8a96cd91b

closed 2017-05-04 by jorgerobles