LaserWeb/LaserWeb4 #258: DXF file contains appropriate layers in qcad; layers showing incorrectly in LW4

PeterGrace, 2017-04-11:
I downloaded this dxf from thingiverse.  I loaded it in QCad and moved the cuts and parts to separate layers to ease the frustration of trying to select a lot of polylines in the LW4 selection box.  Unfortunately upon loading the dxf, the layers in LW4 do not match what QCad shows.

Any insights into why this is happening would be much appreciated!

[Pi-supply_case_rev_3-layers-R14.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/914035/Pi-supply_case_rev_3-layers-R14.dxf.txt)
[IncorrectLayout.json.txt](https://github.com/LaserWeb/LaserWeb4/files/914038/IncorrectLayout.json.txt)

jorgerobles, 2017-04-16:
@iceblu3710

openhardwarecoza, 2017-04-17:
Seconding @iceblu3710

Confirmed the users file in QCad is correct, and further proved that the old parser in LW3 interprets it correctly:

[image]

iceblu3710, 2017-04-18:
PR #266 fixes the issue.

When the layer/entity order is jumbled the layer info was not being carried with the entity when added to the document. byLayer color was also broken in this case. All solved now.

[image]

openhardwarecoza, 2017-04-18:
Thanks @IceBlu3710! Merged PR, it will make its way into tonight's nightly build (:

closed 2017-04-18 by openhardwarecoza