LaserWeb/LaserWeb4 #259: Update package.json with every git commit

openhardwarecoza, 2017-04-12:
Since we display the versions in the app now, mind if we adopt a new methodology and keep package.json updated with every commit ?

Something like  http://stackoverflow.com/questions/13059991/update-package-json-version-automatically

I manually bumped version to 4.0.719 (since we have 719 commits on this repo) as a sort of starting number?

jorgerobles, 2017-04-13:
@openhardwarecoza take a look to https://www.npmjs.com/package/pre-push and 
https://docs.npmjs.com/cli/version

Install prepush and then config to do a `npm --no-git-tag-version version`

I think should work ;)

jorgerobles, 2017-04-16:
Found that (dumb of me) prepush will not work with version unless is a proper commit.
Anyway `npm version` works fine

commit 247ef077 (2017-04-16): Fix #259
closed 2017-04-16 by jorgerobles