LaserWeb/LaserWeb4 #262: Z-Safe Height for milling ops

openhardwarecoza, 2017-04-17:
So, over the weekend I finally build/wired up and commisioned my new mill (provided by OpenBuilds) so at last I am back to hands on testing

Maybe I just dont see it?  But I did take a second look today...   It seems milling ops retract out only to Z=0 between entities, it doesnt go above the surface with some clearance, commonly called Z-Safe height?

[image]

Note: No moves above Z=0 in LW4

In LW3: Note the 10mm Z Clearance between moves - just helps if an edge has a chip up, or when you zeroed it wasnt quite Z=0 

[image]

openhardwarecoza, 2017-04-17:
Oh and it should be configurable per operation - not all op's have the same remaining Z clearance (; 
But a sane default is 3mm 

[image]

tbfleming, 2017-04-17:
Another setting disappeared? I think I called it clearance.

cojarbi, 2017-04-17:
Yes I'm using clearence

openhardwarecoza, 2017-04-17:
Ahh! See even I missed that - may i suggest a clearer label (; 

And as far as I'm concerned that should probably be flagged as required?

[image]

openhardwarecoza, 2017-04-17:
(: Sidenote I want to work on the UI of all these a little: For example in the screenshot, it irritates me that Final Cut Depth and Pass Depth despite being so closely related to each other, are not directly below each other (;

jorgerobles, 2017-04-17:
Huh! I'm showering out glory! Will check :|

openhardwarecoza, 2017-04-17:
No probs Jorge, found it! Needs a better label if it confuses me! #newb @openhardwarecoza go read the #rtfm! Lol!~

cojarbi, 2017-04-17:
So im the only daredevil using the milling?

openhardwarecoza, 2017-04-17:
Probably not - I hope!

openhardwarecoza, 2017-04-18:
https://github.com/LaserWeb/LaserWeb4/commit/31ee365bd421685d4307ec103b4204ff1a0a51c1#diff-34b9ac852e7b8fd2059fcba0ce0c9662R360

closed 2017-04-18 by openhardwarecoza