LaserWeb/LaserWeb4 #263: Purge automatic /docs?

openhardwarecoza, 2017-04-17:
Does anyone ever use it? 
Its not proper documentation, since its automated, and it also isnt really being updated

All it gives me is false positives when I do a project wide search for a phrase, and extra space/bandwidth on git clones (;

tbfleming, 2017-04-17:
Kill it. Unfortunately it won't save git clone time since it's part of the history.

openhardwarecoza, 2017-04-23:
@tbfleming - came across this blog post https://robinwinslow.uk/2013/06/11/dont-ever-commit-binary-files-to-git/ that links to this script https://gist.github.com/nottrobin/5758221 - which claims to be able to rewrite history and get rid of things like we have in this issue, entirely (;

Not saying I want to use it, but figured, since we often get into that "it's now part of the history" problem (like the adobe fonts on dxf a while back) - figured its a good thing to save in case!

tbfleming, 2017-04-23:
I've used that method to clean up others' mistakes at work. It's nasty; you have to coordinate with all other developers and forks. If you don't then either someone loses their recent work, or worse, someone pushes who doesn't understand what happened and all that history comes back in a very nasty merge.

tbfleming, 2017-04-23:
Here's where we stand right now with raw repo size:
```
git clone --single-branch -b dev-es6 --bare git@github.com:LaserWeb/LaserWeb4.git lw4-dev-es6
...
Receiving objects: 100% (4928/4928), 6.23 MiB | 744.00 KiB/s, done.
---
git clone --single-branch -b gh-pages --bare git@github.com:LaserWeb/LaserWeb4.git lw4-gh-pages
...
Receiving objects: 100% (5503/5503), 22.16 MiB | 1.02 MiB/s, done.
```

When it gets too big we can delete the gh-pages branch and create a new one. Note to anyone who does this: gh-pages has the following changes that are *not* in dev-es6. These need to be preserved:
* ```/.gitignore```: doesn't ignore dist/
* ```/webpack.config.js```: removes dev server hooks
* ```/index.html```: browser redirect to dist/

openhardwarecoza, 2017-04-23:
Thats small enough. I bought the 50gb LFS and 50gb bandwidth package for us
anyway, we exhausted the 1gb free bandwidth

tbfleming, 2017-04-23:
It's still giving me this. Their LFS package description is a bit vague; does it cover overall bandwidth, or just bandwidth for LFS files, which we're not currently using?

[image]

openhardwarecoza, 2017-04-23:
Hmm, i clicked the buy more link in the notification that we are over...
Darn will check now

openhardwarecoza, 2017-04-23:
I cant see that error on settings, or or repos? Where do you get it?

tbfleming, 2017-04-23:
https://github.com/orgs/LaserWeb/dashboard

openhardwarecoza, 2017-04-23:
Doesnt show up for me. Maybe one of those notifications that just has to be
cleared? I had that one this morning, clicked the upgrade link. On that
next page Bandwidth was at 1.1gb of 1gb. Punched in Credit card, now it
shows 0 of 50. So i suspect its just the notification?

tbfleming, 2017-04-23:
Just checked on a different device; it's gone. Must be a cache issue.

closed 2017-07-09 by openhardwarecoza