LaserWeb/LaserWeb4 #270: Option to preserve order

tbfleming, 2017-04-18:
There are expert users who manually control the object order in their SVG and DXF files. We already ~~have~~ had an option to bypass clipperjs; ~~we could add an option to bypass our reordering also~~.

had: it was the combine option. I removed it since I solved the original use case another way. ~~We should add it back for this particular use case.~~

On second thought: maybe a totally new option: Preserve Order. This will bypass Clipper and our reordering.

tbfleming, 2017-04-18:
Request from https://plus.google.com/+PetervanderWalt/posts/PJxUmz8KMue

openhardwarecoza, 2017-04-18:
Just a note, but we need to come up with that 'hiding the advanced options' thing discussed in a previous issue, as is we are already a little cluttered sometimes.  Time to start hiding things like this, segment length, etc (all those rarely changed, but good to have on hand options)

tbfleming, 2017-04-18:
This new option can only be on:
* Laser Cut
* Mill Cut

The other ops (inside, pocket, fill, etc.) all need clipper to function.

openhardwarecoza, 2017-04-18:
True, but still, we really need a little cleanup, and thats just one option I want to hide till needed

KirAsh4, 2017-04-26:
I think what I was about to open as an issue is the same as this. When I add several layers to the same operation, the gcode that's generated reverses the order. So for example, if I have an element repeated say 3 times horizontally, I load up each layer into an operation in a specific order, either left to right, right to left, , top-down, whatever ... When I generate the gcode and look at the simulation, it shows that the machine will perform the operation in reverse order than what I put the layers in.

It's not a failure per se as I can easily redo the operation and start with it in the reverse so the gcode will then be generated in the right order, but the order should remain the same as what the user put it in the operation.

jorgerobles, 2017-10-21:
@tbfleming  is this still feasible?

tbfleming, 2017-10-21:
yes