LaserWeb/LaserWeb4 #272: Importing DXF to laserweb4: polyline closed in DXF, not closed in workspace display

PeterGrace, 2017-04-18:
Imported the attached dxf of polyline squares into LaserWeb4 in order to do a cut power/speed test, however the workspace seems to show each polyline missing the bottom part of the square.  I validated the dxf file in inkscape, which shows the squares properly closed.

[CuttingTestWorkspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/930619/CuttingTestWorkspace.json.txt)
[CuttingTestDesign.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/930620/CuttingTestDesign.dxf.txt)

openhardwarecoza, 2017-04-18:
Inkscape makes some rather dirty DXFs. Use qcad, autocad, sketchup etc to
make DXFs. If inkscape is your source, just use SVG

PeterGrace, 2017-04-18:
No, I used QCad to make the dxf, I used inkscape to verify I wasn't crazy. :)

openhardwarecoza, 2017-04-18:
Ah ok (:

tbfleming, 2017-04-18:
Here's another DXF with this issue. The outer path is closed but is importing as open.

[spinner r15.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/931073/spinner.r15.dxf.txt)

This is from https://plus.google.com/102797371787170159448/posts/1h2hRAsf1BX

openhardwarecoza, 2017-04-18:
> 1h2hRAsf1BX

@IceBlu3710

iceblu3710, 2017-04-19:
@PeterGrace the `CuttingTestDesign.dxf` is wonky as the LWPOLYLINE entities are all listed as `shape: false` so my parser skips the last call to close the shape off. I opened your file in DraftSight and saved is at R12 and R14 and only the R12 renders correctly. 

I will need to see if the issue is in the DXF parser itself. [Officially It only supports](http://cncpro.co/index.php/35-documentation/working-with-files/working-with-dxf/28-loading-dxf-files) `R12 ASCII DXF` which is old but reliable.

iceblu3710, 2017-04-19:
PR #273 address the wonky layer grouping that was in the file. 

@PeterGrace According to the [DXF Reference](http://www.autodesk.com/techpubs/autocad/acad2000/dxf/polyline_dxf_06.htm) `70:1` means the shape is closed. You're example file is `70:129` which is not a valid pairing and defaults to `70:0`

PeterGrace, 2017-04-19:
This is a QCad file, as R15 (generated by dxflib) -- let me try R12 and will confirm if that solves the issue.

PeterGrace, 2017-04-19:
Saving the file as R12 (Teigha) in QCad allows the polyline to properly render in LaserWeb4.  I suppose the problem is with QCad then?  As mentioned, Inkscape renders the incorrect dxf as well, leading me to wonder if their definition of the dxf format does not match autocad's reference as you have shown.

Regardless, I'm unblocked; I can save the file in R12.  Thanks for having a look!

iceblu3710, 2017-04-19:
@PeterGrace No problem! 

AutoCad has so many bloody versions and a few in between are not even compatible as they liked to change code pairs meanings but not too unique pairs but to older version ones which makes everything confused. You can see they added a TON of padding to R15 as its literally 10x the file size as R12.

openhardwarecoza, 2017-04-20:
PR #273 merged

closed 2017-05-07 by jorgerobles