LaserWeb/LaserWeb4 #274: Mill ops: S value for spindle speed

openhardwarecoza, 2017-04-20:
From Alex Krause on https://plus.google.com/+PetervanderWalt/posts/PJxUmz8KMue

jorgerobles, 2017-05-04:
Please refine. Is console faulty? Can it be done with macros?

openhardwarecoza, 2017-05-04:
Neg. Should be an option on milling CAM operations: Spindle Speed 0-100%.
Which adds a (S0-100% * s-multiplier) at start of operation.  S pwms the
spindle controller and sets the speed of the endmill spinning. Slower for
alu, faster for wood, for example. Mind you then it also needs to be hooked
to material db

jorgerobles, 2017-05-04:
Ok. Adding option is easy (materialdb not a problem is all synced), Gcode gen maybe harder @tbfleming do you foresee any issue?

tbfleming, 2017-05-04:
Traditionally S value on mills is RPM for compatibility with feed and speed calculators. If I remember right Both grbl and smoothie can adjust their range to calibrate this.

openhardwarecoza, 2017-05-04:
Aha! Correct yes.

jorgerobles, 2017-05-04:
@tbfleming what does imply this apart adding UI option?

jorgerobles, 2017-05-04:
@openhardwarecoza what is s-multiplier? please write gode example of desired scenario. I'm a bit fuzz.

openhardwarecoza, 2017-05-04:
S as in pwm value. S multiplier for smoothieware is 1. For grbl its
whatever $30 is. Its in Settings

jorgerobles, 2017-05-04:
ok, then is the PWM MAX S VALUE on settings?

openhardwarecoza, 2017-05-04:
Yip that (:. Sorry on phone. Will type out example tomorrow

jorgerobles, 2017-05-04:
It's the same as Laser Power, but for milling?

openhardwarecoza, 2017-05-04:
Yip

tbfleming, 2017-05-04:
My recommendation: don't scale S on mills. If the firmware's calibrated correctly, then if the user enters e.g. 4000, and you produce S4000 in gcode, then the spindle will run at 4000 RPM.

jorgerobles, 2017-05-04:
Then is not a percentage. Should be added as an "axis" on machine feed rates?

tbfleming, 2017-05-04:
Yes. e.g. G1 X5 Y4 F100 S4000.

tbfleming, 2017-05-04:
The percentage thing is because lasers repurposed S.

jorgerobles, 2017-05-05:
Done!

closed 2017-05-05 by jorgerobles