LaserWeb/LaserWeb4 #276: Pauses with Smoothieware sometimes fail

openhardwarecoza, 2017-04-20:
from https://plus.google.com/+PetervanderWalt/posts/PJxUmz8KMue 

"When pausing a job it sometimes crashes laserweb (on Smoothie, have to Abort instead)"

domenic-d, 2017-05-03:
Our recent testing has shown that a job will not restart after a pause.
The only way to restart it is to completely abort the job and run it again.

Also, it is not possible to 'abort' a job that is paused. The 'run' button needs to be pressed and then 'abort' will work as expected.

Tested on
Front 4.0.729
Back 4.0.68

sdesalas, 2017-05-03:
FYI
[image]

cprezzi, 2017-05-03:
It was the advise of Jim Morris (wolfmanjm) to use the Grbl compatible commands ! and ~instead of M600 and M601. If they don't work, you probably don't use firmware-cnc and are not in grbl mode.

cprezzi, 2017-05-03:
I have tested it right now with actual firmware-cnc.bin (version edge-8e4a64c) and pause/resume/abort works like a charm.

domenic-d, 2017-05-03:
@cprezzi Can you point me to the discussion regarding the requirement to be running firmware-cnc.bin with with Laserweb?

sdesalas, 2017-05-03:
https://github.com/vlachoudis/bCNC/issues/143#issuecomment-190880616

> wolfmanjm commented on 2 Mar 2016
> You need the very latest edge smoothie, and add a grbl_mode true # to the config.
> 
> The streaming issue is in bCNC and has not yet been fixed AFAIK.

https://github.com/cncjs/cncjs/issues/94#issuecomment-264238719

> tritao commented on 2 Dec 2016 • edited
> Just tried to connect to Smoothieware as suggested by @arthurwolf, with the firmware compiled with CNC=1 which puts it in Grbl-compatible mode.
> 
> The output seems more compatible with what cnc expects, but there's still some problems.

sdesalas, 2017-05-03:
I tried `grbl_mode true` in config on the emblaser2 (Smoothie `edge-67791da5`), I noticed: 

- The `!` command works but lw.comm-server stops issuing wPos events (for tracking X/Y/Z in Jog).
- The '~' command worked ok for continuing.

Didnt notice any other side effects yet but will keep running like this for the rest of the day see if I notice anything else.

domenic-d, 2017-05-04:
@wolfmanjm Thanks for the info Jim. 
Does this mean that if we send a gcode command to move the machine 100mm is one direction, that is it possible to pause this move before it has completed that command by adjusting the line segment?
We were under the impression pausing relied on the emptying of the buffer until it reached the pause instruction.

wolfmanjm, 2017-05-04:
in that case it will complete the 100mm as it is one command being split into 0.5mm segemtns however as the internal planning buffer is full there will be no more commands sent from the host in the queue. That particular case I may be able to pause correctly I'll need to take a look at it.. So I was incorrect.. .with segments on it still means the last gcode still needs to be completed. However the host will still be waiting for the ok from that command. There is of course a downside to segmenting, the cost of some upper speed if the acceleration is slow.

domenic-d, 2017-05-04:
@wolfmanjm Can you please confirm which smoothieware branch we should be compiling and using? We are currently using origin/edge.

cprezzi, 2017-05-04:
@domenic-d We allways communicated to use firmware-cnc and it's also written on our documentation @ http://cncpro.co/index.php/30-documentation/initial-configuration/firmware/smoothieware.

With firmware-cnc the abort (ctrl-x) is immediate, the pause has some delay of the previously buffered commands. I belief we can live with that.

@wolfmanjm Is there an option to connect a pause button to the smootieboard, like the kill button?

wolfmanjm, 2017-05-04:
firmware-cnc is recommended but the only difference is that some of the replies to commands are in GRBL format instead of reprap format, everything else it pretty much the same unless you have a LCD panel in which case he cnc build has a special CNC pendant style LCD readout.

There is no pause in smoothie on button or otherwise. There is a kill button which is useful to immediately kill regardless of how backed up the serial buffers are, especially when using the fast stream mode.

edge is always the branch to use.

domenic-d, 2017-05-04:
Thanks everyone for the info.

closed 2017-05-31 by openhardwarecoza

wolfmanjm, 2017-05-31:
@domenic-d FYI I just added an implementation of the feed hold (!/~) that will pause a long line (100mm) that has segmentation on within the 32 segments in the current buffer. So the answer to your previous question is now yes that will work, even for long lines. (Look for it to be merged soon).

domenic-d, 2017-05-31:
@wolfmanjm That's great!
Can't wait to test this out.