LaserWeb/LaserWeb4 #280: Coreldraw SVGs

openhardwarecoza, 2017-04-20:
Still often have users with issues with files coming from Corel

tbfleming, 2017-04-20:
Is this Corel's randomly-generated px-per-inch value, or are there other issues also?

FearNaBoinne, 2017-04-24:
I have many SVGs from Corel that don't even show up in the viewport area when opened in LW4

openhardwarecoza, 2017-04-24:
So, post them!

jorgerobles, 2017-04-24:
@FearNaBoinne Please follow the requirements https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/.github/ISSUE_TEMPLATE.md

FearNaBoinne, 2017-04-24:
I'm already working on it guys, but I also have a job that requires attention!  :)

FearNaBoinne, 2017-04-24:
Here's one that doesn't open at all (and no errors or messages are provided)

[Peeters CoA BW (small).zip](https://github.com/LaserWeb/LaserWeb4/files/951026/Peeters.CoA.BW.small.zip)

Not one of the files I was looking for (those open, but just don't show up at all in the workspace), but a problem Corel SVG none the less...

FearNaBoinne, 2017-04-24:
>> Ok, testing is interrupted on my side for now, as I installed 4.0.724, and now nothing can be opened anymore! <<

openhardwarecoza, 2017-04-24:
[image]

opens fine for me... Relax with the attitude a little (; - its all nightly builds anyway, just got to the office, busy uploading backend 4.0.68 with 7.0.724 frontend builds.

FearNaBoinne, 2017-04-24:
Just letting you know I cannot test any further right now... I am willing to invest as much time as I can in to helping y'all in getting it to work, but between work, a family and a now broken system, there's just so much I can do!  :P 
Will test the new BE version ASAP and if necessary do a full wipe of all the LW related folders, etc. to see whether that helps!

openhardwarecoza, 2017-04-24:
https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/v4.0.742-68 is available

No need to wipe.  PS: Dont recognise the Nickname.  Are you on G+?  Feels like i am talking on two platforms about the same issue now

jorgerobles, 2017-04-24:
I've scoped your file error. Gonna dig thru this

jorgerobles, 2017-04-24:
@openhardwarecoza parser throws `no supported tags error`. Could you attach your file? did you open on inkscape or so before?

jorgerobles, 2017-04-24:
[image]

Hmmm AI cannot load the file either O_o

jorgerobles, 2017-04-24:
Inkscape ..
[image]

I feel that bitmap fills are not right, isn't it?

jorgerobles, 2017-04-24:
after inkscape:

[image]

FearNaBoinne, 2017-04-24:
Peter, I am CescoAiel on G+

Jorge: 
It opens in InkScape, but I did not make any changes to it in there on purpose, as that is what Corel spit out... 

And the bitmap fills are indeed screwed up in Inkscape... I noticed that in InkScape myself as well, but again left it as is, because of the fact we're lookingat Corel SVGs not opening properly!

Zipped up the CDR for you:
[Peeters CoA BW (small).zip](https://github.com/LaserWeb/LaserWeb4/files/951890/Peeters.CoA.BW.small.zip)

jorgerobles, 2017-04-24:
> Zipped up the CDR for you:
Peeters CoA BW (small).zip

LOL. I'm on Mac.

FearNaBoinne, 2017-04-24:
Macs can do zip too!  I'd upload as a CDR, but github won't let me...

openhardwarecoza, 2017-04-24:
Ahh see thought i was feeling like I repeat myself! (if you use a nickname,
put your real name in real name so your brownie points and earned respects
can count for something. Otherwise we think its just some random new guy
lol)

jorgerobles, 2017-04-24:
>Macs can do zip too! I'd upload as a CDR, but github won't let me...
@FearNaBoinne :rofl:  Got no Corel on Mac.

FearNaBoinne, 2017-04-24:
BTW @openhardwarecoza: The 4.0.68 installer still reports BE 4.0.63!
[image]

tbfleming, 2017-04-24:
Chrome also reports the file is corrupt. To test a .svg file using Chrome, drag it into an empty tab. Weird part: the .svg file uses but doesn't declare the sodipodi namespace. sodipodi is the predecessor to Inkscape; no other tool should mess with the sodipodi namespace. Corel messed up big time.

That's 3 strikes against this file:
* Chrome reports it's corrupt
* https://validator.w3.org/check reports it's corrupt
* Inkscape mishandles it

FearNaBoinne, 2017-04-24:
Gotta love Corel, eh?...

Once I got LW4 working again, I'll find other files to test with! This one isn't the greatest file to work with!

openhardwarecoza, 2017-04-24:
Best to always start off with files you draw yourself.  Then theres no
"garbage from the 10x it was converted from the original guy who drew it in
1992" legacy

tbfleming, 2017-04-24:
It now makes sense. Chinese laser boards love Corel because they have the same quality control :)

jorgerobles, 2017-04-30:
Closing for now...

closed 2017-04-30 by jorgerobles