LaserWeb/LaserWeb4 #282: Change position of initial z axis travel move

domenic-d, 2017-04-21:
We have noticed that the initial G0 z-axis move is occurring in the wrong place in the generated GCode.

Right now, the initial G0 z-height is being set before the initial G0 x & y moves. This could potentially lead to a collision with material.

**Current GCode**
```
; Pass 0
; Pass Z Height 10mm (Offset: 0mm)
G0 Z10.00
; Pass 0 Path 0
G0 X148.70 Y92.78
```

**Improved GCode**
```
; Pass 0
; Pass 0 Path 0
G0 X148.70 Y92.78
; Pass Z Height 10mm (Offset: 0mm)
G0 Z10.00
```

Note: The ending GCode is correct as it performs the final G0 z-axis move before the final G0 x & y moves.

tbfleming, 2017-04-21:
Is there a clearance Z value like milling machines and routers have? Here's the sequence for those:

* Move to clearance Z
* Move to starting XY
* Move to starting Z

domenic-d, 2017-04-21:
Our z-clearance position is the laser head in the highest position. This is where the laser head homes to and also returns to after the completion of a job.

To keep everything 'generalised' then a z-clearance height in the settings when z-axis is enabled would work best.

Then the sequence you suggest would work the same as for mills/routers etc.

jorgerobles, 2017-05-07:
As for now, the way the gcode is generated have no way to achieve that exception. That would lead to a bunch of hooks for "after the first pass first path but only once" bloating the code or a huge refactor.

The only way I could think of is use of macros hooks available on operations. onJobStart, going to Z clearance position will suffice.

domenic-d, 2017-05-07:
Maybe this request has become confused.
As in the example I have supplied in my initial post, the first z-move should happen after the first xy-move.

It is not an exception, it is a re-ordering of when the first z-move happens.

jorgerobles, 2017-05-07:
Yes, @domenic-d, I didn't mean that "exception" to be unnecessary.
It's an exception to how the actual gcode generation code is done (implying some deep loops), and cannot solve without a huge refactor. Maybe @tbfleming could inspect it further to find the best place to insert.

domenic-d, 2017-05-07:
I expect this is absolutely imperative for milling. I haven't checked whether this is already implemented in that side of laserweb, but i would be surprised if it isn't.

This is perhaps another good example where a post-processor infrastructure would have been very useful.

jorgerobles, 2017-05-07:
@domenic-d  Mill does it indeed. I've hacked to use on laser cut operations. Hacking on raster will be revised later.
Please test thoroughly. Maybe does not behave on all cases.

jorgerobles, 2017-05-07:
@openhardwarecoza, @tbfleming, @cprezzi, @domenic-d 
I've been looking for a postprocessor spec to not to reinvent the wheel but have not found anything merely structured. 
The simplest engine could be done using a web-worker that receives all the core gcode.
Is up to the post contributor to code whatever he wants. Should be posted on a centralised repository like lw.machines to be approved as a PR. LW could expose those file names depicted on a JSON manifest file. The most LW can actually expose is some comments with a predefined format to hint the postprocessor what's going on core process. I would appreciate concrete implementation ideas :smile:

tbfleming, 2017-05-08:
Probably should make that a separate issue. First thing to nail down: what are users trying to accomplish with postprocessors?

jorgerobles, 2017-05-08:
@domenic-d ?

domenic-d, 2017-05-08:
@jorgerobles Initial z positioning now looks correct.
Thank you.
I will close this off and let the core team decide how they wish to proceed with a post-processor, if at all.

closed 2017-05-08 by domenic-d