LaserWeb/LaserWeb4 #283: Migration of macros to settings redux space

jorgerobles, 2017-04-22:
Hi all @LaserWeb/devteam 

I've opened a branch (https://github.com/LaserWeb/LaserWeb4/tree/macros_settings) migrating macros to settings, according with @cprezzi suggestions

This allows to save the `macros` along the `machine profiles`, and supporting vendors to introduce their own macros on `lw.machines`

It's a somewhat hairy issue, migration is done but one-way (cannot use prior versions unless reset localstorage)

Please test and report.

jorgerobles, 2017-04-23:
@cprezzi have you configured smoothie to put the pwm_period_ms with Gcode?

jorgerobles, 2017-04-23:
@cprezzi, alternatively as your comment at https://plus.google.com/117194752728291709572/posts/aHfK8uYvi3w

I can just attach fixed (only available thru `lw.machines `file edit) macros. That macros will be inherited using the `*generic_grbl `or `*generic_smoothie` profiles and then saved as new machine profile. Maybe this option is closer to the idea expressed on G+

cprezzi, 2017-04-24:
@jorgerobles I just tried to read the "laser_module_pwm_period" via serial, but don't get any response. A query for accelleration or x_axis_max_speed is working.

closed 2017-04-26 by jorgerobles