LaserWeb/LaserWeb4 #284: Workspace import doesnt honor mm/s vs mm/min setting

openhardwarecoza, 2017-04-24:
### Follow this steps to open an issue ###

- [x] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

[4040foot.zip](https://github.com/LaserWeb/LaserWeb4/files/951032/4040foot.zip)
[4040foot dxf.zip](https://github.com/LaserWeb/LaserWeb4/files/951037/4040foot.dxf.zip)

I created this workspace on my home PC where settings are on mm/sec 

Took it to PC in the garage, which, i didnt notice, was on mm/min (; 

Loading the workspace loaded my 150mm/sec as 150mm/min - and since I didnt notice, i merilly generated gcode, and ran... luckily very much slower than planned - had it been the other way round i wouldve needed a new endmill and spare underpants (; 

I guess best practice it should either always export in mm/min (and be changed to mm/sec if loaded with that setting enabled) or something like that (;

jorgerobles, 2017-04-24:
@openhardwarecoza workspace snapshot captures 'documents', 'operations', 'currentOperation' states to download. Could include `settings` but that could lead unwanted behaviour, as is not strictly `workspace`  😏 
Other way to achieve that is using tools application snapshot that makes a copy of the specified state keys.  Being "mm/s" setting critical, could try add subkeys saving and add it to workspace snapshot ('settings.toolFeedUnits')

openhardwarecoza, 2017-04-24:
Or perhaps we could always capture as a known unit (mm/min as thats what we
send the firmware anyway) , and convert everywhere its displayed as mm/s?
Ie not a change to workspace export, just a change to the way values are
handled? (mm/min under the hood, /60 whenever it needs to be?

jorgerobles, 2017-04-24:
.. :smile: ugh, maybe lot of work that way. Have to search whole application for the use of that setting in each different team implementation. Wanna help?

openhardwarecoza, 2017-04-24:
Hehe Yeah leave it open for now will do soon! Its not a majorly urgent one

jorgerobles, 2017-05-04:
We should ditch mm/s and done 😆

openhardwarecoza, 2017-05-04:
Lol i kinda agree. Its the corel/laserdrw crowd that didnt want to

openhardwarecoza, 2017-05-04:
We can always force it on them

FabCreator, 2017-05-04:
Or ditch mm/min and so for SI units only ;)

cprezzi, 2017-05-04:
For me it's easyer to imagine mm/s, but it looks like there is no real industry standard (see https://en.wikipedia.org/wiki/Speeds_and_feeds).

openhardwarecoza, 2017-05-04:
(: the Mericans want inch per minute and G20 support (;

cprezzi, 2017-05-04:
Not all of them ;) The US industry is slowly moving :)

openhardwarecoza, 2017-05-04:
(; ok let me re-stereotype lol. The old ones lol

FabCreator, 2017-05-04:
I agree with @cprezzi i find mm/s easier to imagine, too many zeros for mm/min. 100mm/s vs 6000mm/m
Lets do the Americans a favour and leave out imperial from laserweb ;)

jorgerobles, 2017-05-04:
With patch #284, workspace snapshot (which is indeed a part of the whole application state) stores also settings.toolFeedUnits. So, if you load that workspace, that setting will overwrite config. That fixes this issue, but caveats are that as is done, that setting change is permanent upon workspace load.

An alternative way is to store the units with the value (Ex: "100 mm/s" ) then the loader and inputs have to resolve (regexp) this kind of notation. That's an UI challenge then.

tbfleming, 2017-05-04:
jscut stored units with values; it created a large source of bugs that I had to fix. It also forced me to roll back a pull request I accepted a few months later because that pull request interacted subtly with units creating a hard-to-reproduce bug that a few key users kept hitting. Units with values did have a user advantage though; users could enter imperial tool units and metric operation units without having to do the conversion on their own.

jorgerobles, 2017-05-04:
Ok then. Will figure out a workaround to settings overload for the patch
and done! :)

jorgerobles, 2017-05-04:
Well, cannot figure a more elegant way that is currently done :/ Closing for now.

closed 2017-05-04 by jorgerobles