LaserWeb/LaserWeb4 #285: Default macro's are now locked, but still not wanted...

FearNaBoinne, 2017-04-24:
The [f1] Goto XY zero and [f2] LASER OFF macro's are somehow now locked macro's (4.0.724), but I still don't want them there!

Even though initially I *was* able to delete them when I was setting up 4.0.724 (As it somehow lost all my prior settings on upgrade), they came back after the first restart of LW4.

Why lock macro's without any way to unlock them?

[laserweb-snapshot.json.txt](https://github.com/LaserWeb/LaserWeb4/files/951757/laserweb-snapshot.json.txt)

jorgerobles, 2017-04-24:
Locked macros are shipped with LW, and are not removable. Are meant to be vendor shipped. Could work for a solution on this, thought. @openhardwarecoza, @cprezzi, any ideas?

FearNaBoinne, 2017-04-24:
It kind-of screws up my process, because I already have things assigned to f1 and f2, and neither of the 2 predefined ones would ever get used by me! 
However, if it's too much of a challenge at this time, I guess I can work around it...

(The FABkit for instance homes to Xmin Ymax, and there is 0 reason to want to move the head to Xmin Ymin, because it'd only be in the way there!)

openhardwarecoza, 2017-04-24:
Hmmm, i see the point why they are locked, IF its a vendor profile (Dont
want users deleting a vendors macro as then when the vendor says "click
that" they say "I dont have that button"

But i guess we can assume the non-vendor default profile just has NO
predefined macros then?

cprezzi, 2017-04-24:
What about give the predefined macros a "locked" flag?
If set, then user cannot change it, if not set, user can change and delete.

cprezzi, 2017-04-24:
This way, we would be able to deliver usefull preset marcros, but only some of them are locked because the vendor needs them.

cprezzi, 2017-04-24:
A complete different approach would be, that we create an online database for machine profiles and macros, that can be downloaded on users liking, but are not auto-installed with the setup.

jorgerobles, 2017-04-24:
So, rounding up :)

- Macros should have a locked attribute (already have one, and currently is applied to any bundled macro).
- User defined macros could not be locked.
- Default bundled profiles _should_ expose macros that are not locked. 
- Vendor bundled profile can expose macros either locked or unlocked.
- The only way to restore a deleted bundled macro is to reapply the profile.

cprezzi, 2017-04-24:
@jorgerobles Sounds good to me, but I wouldn't make a difference between default and vendor bundled profiles, in case of locked or unlocked macros. Default bundled profiles can also have both types of maros. For example: If we want to clean up the Jog buttons and move the LasterTest button to the macros area, but want this macro to be locked.

jorgerobles, 2017-04-24:
@cprezzi yes that was the **should** :)

FearNaBoinne, 2017-04-24:
That all sounds like a great improvement, but I would really like if there was a possibility to at least change the hotkeys for the macro's, so I'm not forced to work around whatever hotkeys are selected... That way you can still enforce the presence of certain vendor macro's, but allow the user to select his preferred hotkeys...

jorgerobles, 2017-04-24:
We could strip keybindings from bundled macros, for now. Reordering will be done by "the rule of three": at least three people ask for it :D

cprezzi, 2017-04-24:
@FearNaBoinne Not my opinion. Locked macros should also have locked shortcuts. Otherwise the vendor can not tell the user to "just press ctrl-s"... 

It's absolutely usual that softwares have thier own shortcuts. We also have reserved shortcuts for other functions, like jogging, so the user is anyways not free to define what he want's. I think it's better to have standard shurtcuts for all our users that they can get used to.

cprezzi, 2017-04-24:
Probably we could try to use alt-key combination for internal shortcuts and ctrl-key for user shortcuts? Or what ever is convenient.

FearNaBoinne, 2017-04-24:
For *me* (but that's only me, I cannot speak for other users) the most important bit would be to keep the Fn keys free... Other than that, possibly the alt/ctrl number keys, but absolutely the Fn keys...

cprezzi, 2017-04-24:
@FearNaBoinne That sounds reasonable to me :)

jorgerobles, 2017-04-27:
Done on .726

closed 2017-04-27 by jorgerobles