LaserWeb/LaserWeb4 #286: Macro Settings GUI not updating correctly (DELETE button)

FearNaBoinne, 2017-04-24:
Once you've hit a locked macro in the settings - macro pane, and the DELETE button gets disabled in the GUI (4.0.724), it stays disabled, even if you then select an unlocked macro.

[laserweb-snapshot.json.txt](https://github.com/LaserWeb/LaserWeb4/files/951763/laserweb-snapshot.json.txt)
The above file has several macros, and only 2 are locked (even though I don't want them to be, but that is a different issue), but once you select one of the 2 locked ones (at the bottom) none of the macros can be deleted anymore...

jorgerobles, 2017-04-24:
I'm aware. Will patch ASAP.
There's a macros code on its way, moving to machine profiles. This also fix this issue

jorgerobles, 2017-04-24:
Moving Macro behavior discussion to #285

closed 2017-04-24 by jorgerobles