LaserWeb/LaserWeb4 #289: REQUEST: Clear-all on operations

KirAsh4, 2017-04-26:
While one can remove an image or operation by clicking on the red [X] on each line, when you have multiple operations, it gets tedious to be clicking on each line to remove them. Can there be a single 'remove all' button to clear everything (similar to the trashcan icon to clear the visible gcode from the canvas)? Doesn't matter much on the loaded images, but more so for the operations. Some of my designs will have quite a lot of operations because I switch between engraving, marking, and cutting all on the same design, and I do it in a specific order as I traverse the material (to avoid things like loose pieces getting in the way of the moving head, or if the board is warped, I tend to do all engraving and inner cuts in a specific order before doing the final large cuts.)

tbfleming, 2017-04-26:
Do you mean remove all operations, or move all document objects within a single operation?

KirAsh4, 2017-04-26:
I suppose both, though in this case I'm referring specifically to removing all operations.
[image]

openhardwarecoza, 2017-04-26:
Clicking on the document level, deletes all children. Same for layer level

KirAsh4, 2017-04-26:
That's not what I'm asking for.

openhardwarecoza, 2017-04-26:
Exactly what your first post + screenshot did ask

openhardwarecoza, 2017-04-26:
Except for the operations (note i mentioned Document level)

KirAsh4, 2017-04-26:
My first post, and the screen shot, talks about the operations, clearly defined and noted in the image as well. Nowhere do I say that I want to get rid of the (multiple) layers within an operation.

KirAsh4, 2017-04-26:
Even Todd asked, 'delete all OPERATIONS' or all documents. Again, nowhere does it talk about the individual layers.

openhardwarecoza, 2017-04-26:
<thumbsup>

jorgerobles, 2017-04-28:
Done. Will show next build.

closed 2017-04-28 by jorgerobles