LaserWeb/LaserWeb4 #290: REQUEST: Re-order multi-layer operations

KirAsh4, 2017-04-26:
I recently noticed that when I load up several layers in a single operation, that the generated gcode is in reverse order. That's, I believe, issue #270 ... The only option right now is to delete that operation, and recreate it with the layers in the reverse order. As with re-arranging actual operations, can there be an option to do that with the individual layers within an operation as well so there isn't a need to delete and recreate it again, but simply push a layer up or down like you can do with operations.

tbfleming, 2017-04-26:
I'm leaning against this since it adds more buttons to the UI and there's already a way to handle it: use multiple operations.

jorgerobles, 2017-04-26:
Secondly @tbfleming. Better use operations. As far as is no way to order
inner to outer paths, splitting on operations does all what you need.

KirAsh4, 2017-04-26:
Right, that is what I'm doing now, create separate operations, but as some of my designs end up with a long list of operations, I started playing with grouping several together and that's when I discovered the reverse order and thought, what if there's a way to re-order them. Of course, if the order is maintained to begin with, then the ability to re-arrange becomes more of a moot point.

closed 2017-05-04 by jorgerobles