LaserWeb/LaserWeb4 #293: FileField: nullify's setTimeout fires before handleChange

tbfleming, 2017-04-27:
This is happening on @FearNaBoinne 's Pipo:
* FileField.handleClick
* focus event
* setTimeout handler
* FileField.handleChange

Do we need the code in the setTimeout? It creates a race condition.

jorgerobles, 2017-04-27:
There's no other way to handle cancelling a file chooser dialog. 
If don't use this pattern, then react misfires several times the click event causing load misbehavior, and losing first click event, as react reuses events. :|

FearNaBoinne, 2017-04-27:
Is there a way to set a flag to indicate when it is safe to delete the handler? Then you could reset the timer as long as it isn't safe yet, and delete the handler when it is safe to do so...

jorgerobles, 2017-04-27:
It's safe when the file has changed or loaded, but as there's none telling the file has not been loaded (by cancelling) that's the only way I found...
Prior way to do was creating dom input ad-hoc, but was error prone too (missed 25% events)

FearNaBoinne, 2017-04-27:
But on my Pipo it apparently does it between me clicking the add document button and me clicking the OK button after selecting a file, so something's not going quite right there!...  :o

FearNaBoinne, 2017-04-29:
Don't get me wrong, I'm **_not_** being impatient nor do I want to rush y'all, _just trying to figure out the best way forward_ (which might actually be backward) for now, as I need to get the machine working for my wife (she's got a few small projects pending):

Any ETA on a fix for this one?

FearNaBoinne, 2017-04-29:
I can give you remote access to mine for testing, just remember it's connected (hardwired and screwed down) to my FABkit laser! and please don't use it to hack my network!... ;)

FearNaBoinne, 2017-04-29:
(It's weird tho that there is no cancel event being triggered... It's a serious oversight in the design there!)

jorgerobles, 2017-04-29:
> (It's weird tho that there is no cancel event being triggered... It's a serious oversight in the design there!)

Yes.. Give me the weekend. Maybe can find something.

FearNaBoinne, 2017-04-29:
Thanks and good luck, but that means I will have to revert to the old build for now, so she can at least use the machine! 
I've got the settings saved in localstorage and export file, so it's easy enough to get those back once the file-issue is resolved!
(I'll also do a few donations this weekend, salary's in! <G>)

jorgerobles, 2017-04-29:
Beware! Recent macros rework could render localstorage broken. Please export your settings from settings tools. We can work a solution only if you have a backup

jorgerobles, 2017-04-29:
@openhardwarecoza 🍺 and 🍕!

FearNaBoinne, 2017-04-29:
To the above two: 
1) Notice the "and eport file"  ;)
2) Yes, indeed!

jorgerobles, 2017-04-29:
Careful man worths double.

openhardwarecoza, 2017-04-29:
> Hi.. I was right now on it :)
> [image: image]
> <https://cloud.githubusercontent.com/assets/1706080/25553778/f0ba434e-2cbc-11e7-8d33-c3cb4a8d7e16.png>
>
> But no ETA sorry. Is a very specific problem to navigator on Pipo. This is
> the most used feature (load files) and only your issue has arised for now.
> I would need a Pipo.
>
> I'm trying to figure something, but no promises. FYI:
> https://www.google.es/search?q=filefield+catch+cance&oq=
> filefield+catch+cance&aqs=chrome..69i57.6863j0j7&
> sourceid=chrome&ie=UTF-8#safe=off&q=file+field+catch+cancel

(; if only we could get a Pipo to @jorgerobles (; hint hint @FabCreator and
its customers. Would probably get a touch interface going sooner too (;

FearNaBoinne, 2017-04-29:
Well Mr @openhardwarecoza, if you can at least get that donate link in your github working, I may not be able to afford a Pipo to donate to the good cause, but I would be able to increase the beer and pizza fund a tiny bit!

openhardwarecoza, 2017-04-29:
(: i am not allowed any more donations actually (because of conflict with
dayjob) but all the devs links are on the About page inside Lw(;

FearNaBoinne, 2017-04-29:
Ah, ok! I'll do the direct thing then!

jorgerobles, 2017-04-29:
@FearNaBoinne, @openhardwarecoza It should work now. No more timeouts nor debouncing.

jorgerobles, 2017-04-29:
Will compile new version on two hours or so.

FearNaBoinne, 2017-04-29:
You, sir, rock!  👍 💯 🎉

jorgerobles, 2017-04-29:
https://github.com/LaserWeb/LaserWeb4/releases/tag/v4.0.729

FabCreator, 2017-04-29:
Can definitely look into getting a Pipo to a dev willing to help keep laserweb running smoothly on it.
@jorgerobles can we tempt you?

jorgerobles, 2017-04-29:
Thank you but I'm full hands right now, sorry :|

openhardwarecoza, 2017-04-29:
@cprezzi - to get that minimal touch interface?

FearNaBoinne, 2017-04-30:
Per @jorgerobles request, Snapshot of my config on the Pipo

[laserweb-snapshot-170430.zip](https://github.com/LaserWeb/LaserWeb4/files/966663/laserweb-snapshot-170430.zip)

jorgerobles, 2017-04-30:
Thanks.
As fast workaround, please rebuild your macros. That is, delete them from macros panel. Close and restart app. Rebuild macros on app.
Here's your config portion to copy/paste.
Will check macros format migration ASAP
<pre>
"macros": {
      "f1": {
        "label": "Autofocus",
        "gcode": "G91                       ;Relative positioning\nG0 X11 Y-13.5     ;Move 11mm right and 13.5 mm forward\nG28.2Z                 ;Home Z\nG0 X-11 Y13.5     ;Move 11mm left and 13.5 mm backward\nG90                       ;Absolute positioning\n"
      },
      "ctrl+f1": {
        "label": "Autofocus -5mm (Yellow)",
        "gcode": "G91                       ;Relative positioning\nG0 X11 Y-13.5     ;Move 11mm right and 13.5 mm forward\nG28.2Z                 ;Home Z\nG0 X-11 Y13.5 Z-5 ;Move 11mm left,13.5 mm backward and 5mm down\nG90                       ;Absolute positioning\n"
      },
      "f2": {"label": "Home XY", "gcode": "G28.2XY\n"},
      "f4": {"label": "Motors OFF", "gcode": "M18\n"},
      "g5": {"label": "LASER OFF", "gcode": "M5"},
      "f12": {
        "label": "RESET Smoothie",
        "gcode": "M80  ; SmoothieRST  has been wired to 0.25, which will be pulled down by the M80 command\n"
      }
    }

</pre>

FearNaBoinne, 2017-04-30:
I've got them stored in a text file, but thanks anyway!  ;)

I'll test and let you know!

FearNaBoinne, 2017-04-30:
That indeed fixed the keys/typing problem!
Thanks again!

(But I still cannot add document to operations...  😞)

jorgerobles, 2017-04-30:
Are you adding with drag or buttons?
Also, Please goto settings, tools and enable dev tools. Please report if any error is on console. Thanks!

jorgerobles, 2017-04-30:
@FearNaBoinne found material database incompatibility. Regretly, we cannot afford backwards compatibility. Your version is before material database refactoring couple of months ago. 
Steps to fix.
Fast workaround:

1. Go to settings, open dev tools -> application, left pane localstorage. Delete
[image]
2. Restart application.
3. Edit you config file, strip material database portion (and macros if so). Load config.

jorgerobles, 2017-04-30:
@tbfleming have to find a way to catch component exceptions on instancing. It's not easy to achieve backwards compatibility in some cases. For now, I will add an application reset button on about pane.

jorgerobles, 2017-04-30:
@openhardwarecoza, @cprezzi, @tbfleming, moved reload window, dev tools and added Reset to factory, now available to about tab. Easier to find.

closed 2017-04-30 by jorgerobles