LaserWeb/LaserWeb4 #294: Warning for low feed-rates

domenic-d, 2017-04-28:
We have had a situation recently where a user has accidentally set a very slow feed-rate (10mm/min) in their machines and run the job. It caused their material to ignite but luckily was picked up in time.

At such slow speeds the potential for ignition of the material goes us significantly and although jobs should always be monitored, sometimes they just aren't.

For our machine, running it anywhere below 150mm/min is not advised.

I wanted to ask for thoughts on whether a warning is raised if feed-rates below a certain threshold are used?

In light of so many new users working with LW4 every day, I believe this is something worth discussing.

openhardwarecoza, 2017-04-28:
The hard thing is, what about my little 100mw cd drive laser? Or my cnc
mill, chowing slowly through steel? Sometimes low rates are a thing. So
we'll need to hook it onto vendor related info too.

Maybe, the better way to prevent a user from doing so, is to flesh out the
material DB for them before shipping with a lot of materials?

domenic-d, 2017-04-28:
Yes, it is definitely machine specific and probably more important for lasers than mills.

As much as I am against adding extra options to an already option-rich UI, a field in the machine profile with a threshold-alert could do the trick. By default it is empty. A value entered would see a dialogue box open that gave the user a warning:
Warning: Your Cut-Rate is below the recommended value for your machine. This could be dangerous...etc.

openhardwarecoza, 2017-04-28:
Yip, i do like that plan. Min and max rates would be cool.

Then vendors with flimsy frames can set a speed limit at the top too.

domenic-d, 2017-04-28:
Yes, I didn't consider a max value, but you are absolutely right.

openhardwarecoza, 2017-04-28:
Approved (; we need this!

jorgerobles, 2017-04-28:
<pre>
machineFeedRange: {
        X: {min: 0, max:10000},
        Y: {min: 0, max:10000},
        Z: {min: 0, max:10000},
    },
</pre>

domenic-d, 2017-04-28:
Warning: You have set a cut rate (#value) outside the limits of your machine (#min - #max). We recommend you review your settings to prevent a potential hazard.

jorgerobles, 2017-04-28:
@openhardwarecoza where would it be? a validator on cutrate operation?

openhardwarecoza, 2017-04-28:
Jip (: that should be easiest (red flag on the side is more consistent than
the popup idea)

jorgerobles, 2017-04-28:
@openhardwarecoza ... maybe not correct. as I understood that has to be checking axes (as my former comment, and yes, xy could be put together)

Should we use an overall feed setting or by axis?

openhardwarecoza, 2017-04-28:
Hmmm valid point, z may differ, and z is always slower than xy. So should
we need max/min settings for xy and z seperate?

openhardwarecoza, 2017-04-28:
(in cam, z moves are the plunge rate)

jorgerobles, 2017-04-28:
And what about material database? should be restricted on selected profile? (Could be lead to invalid data if we delete a profile ... ugh)

cprezzi, 2017-04-28:
If we only place the warning in CAM, we will miss all direct loaded gcode!
It would be save to do a check on `run` sommand.

jorgerobles, 2017-04-28:
@cprezzi same idea!

jorgerobles, 2017-04-28:
Seems the gcode parser has some things to do :P

jorgerobles, 2017-04-28:
And should dispatch error flag to COM redux to avoid firing O_o

jorgerobles, 2017-04-28:
Done. I've done tests, but please check before compilation. As for now, It warns on operation, but not on jog.

KirAsh4, 2017-04-30:
Referring to #296:

When I still had the original stock M2 controller in, I was pushing 400mm/s however that was simply turning on/off, no PWM control like now. So I completely understand a slower rate because of a lot more information. I was just curious why 10000mm/min is the upper limit now.

For small pieces (say less then 10cm across), I will run between 150mm/s to 200mm/s (9000-12000mm/min). On larger pieces where it traverses the full 12" width of the sheet, I'll bump it up to 250mm/s (15000mm/s) or depending on the **simplicity** of the image, up to 300mm/s (18000mm/min).

tbfleming, 2017-04-30:
There is no sane max for all machines. Maybe default max=1e100 (effectively no max).

jorgerobles, 2017-04-30:
@KirAsh4, my fault. :P  . I did such arbitrary feed rate. I've currently disabled maximum feed rate (available on next build)
@tbfleming Infinity will suffice :D

jorgerobles, 2017-04-30:
Also, there's a pending job to refactor LW to store only mm/min but visualizing mm/m and mm/s---

KirAsh4, 2017-04-30:
@jorgerobles, please don't misunderstand, I'm not against the limitation. I was simply curious whether there was some underlying reason I wasn't seeing. 10,000mm/m is a nice, round number to work with. :)

jorgerobles, 2017-04-30:
Thanks, but as @tbfleming cunningly said, I realized that limitation is wrong to be hardcoded. :)

domenic-d, 2017-05-02:
Working well. Everyone happy to close this off?

KirAsh4, 2017-05-02:
Before you close this, has there been an update to the current rate limitation? Looks the binaries haven't been updated in the last 10 days, assuming there has been an update.

jorgerobles, 2017-05-03:
https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/v4.0.730-68

closed 2017-05-03 by jorgerobles