LaserWeb/LaserWeb4 #295: Missing JSON files prevent compilation

hepabolu, 2017-04-28:
```
git clone https://github.com/LaserWeb/LaserWeb4.git   # dev_es6, 4.0.728
cd LaserWeb4
npm install
npm start
```

Fails, because 2 files are missing:

ERROR in ./src/reducers/machine-profiles.js
Module not found: Error: Cannot resolve 'file' or 'directory' ../data/lw.machines/machine-profiles.json in [path]/LaserWeb4/src/reducers
 @ ./src/reducers/machine-profiles.js 20:76-128

ERROR in ./src/reducers/material-database.js
Module not found: Error: Cannot resolve 'file' or 'directory' ../data/lw.materials/material-database.json in [path]/LaserWeb4/src/reducers
 @ ./src/reducers/material-database.js 34:66-120

After adding both files with an empty JSON object compilation succeeds

jorgerobles, 2017-04-28:
<pre>
git clone https://github.com/LaserWeb/LaserWeb4.git   # dev_es6, 4.0.728
cd LaserWeb4
npm install
</pre>
**`npm run installdev`**
<pre>
npm start
</pre>

closed 2017-04-28 by jorgerobles

hepabolu, 2017-04-28:
Maybe good idea to note this in the README ? I was just following instructions in the README.

jorgerobles, 2017-04-28:
Will do, sorry

openhardwarecoza, 2017-04-28:
@hepabolu are you a user or a dev?

Devs are usually familiar with checking package.json for scripts. The fact
you mentioned following the manual, makes me want to check if you arent a
user trying to install LW, in which case, head over to
github.com/LaserWeb/LaserWeb4-Binaries instead (:

jorgerobles, 2017-04-28:
@hepabolu please, what readme are you following? Thanks!

openhardwarecoza, 2017-04-28:
(; good point @jorgerobles we dont have any instructions in the LW4 readme,
so its not from that he was following

hepabolu, 2017-04-28:
@jorgerobles The readme found here, in plain sight: https://github.com/LaserWeb/LaserWeb4
@openhardwarecoza I'm a programmer, but not good enough in NodeJS to help out with this project, so in this case I'm more of a user. I installed this project, because there are currently no OSX LaserWeb4 binaries and I wanted to test if the macro buttons work in 4.0728. So I simply followed the instructions on the front page of this project. ;-)

openhardwarecoza, 2017-04-28:
There is a dmg on the latest release

hepabolu, 2017-04-28:
???? On https://github.com/LaserWeb/LaserWeb4-Binaries there is a message that there is no dmg. Where can I find the dmg ?

openhardwarecoza, 2017-04-28:
One question mark would suffice. Under the releases tab