LaserWeb/LaserWeb4 #297: GCode circles not displayed correctly

domenic-d, 2017-05-02:
Imported gcode of a circle is not displayed correctly in the workspace.
Job runs correctly.

LW version: 4.0.729
Workspace not attached because gcode is not saved within it.
Gcode file attached.

[image]
[circle.nc.zip](https://github.com/LaserWeb/LaserWeb4/files/972186/circle.nc.zip)

tbfleming, 2017-05-03:
Known issue. The temp gcode parser doesn't support most of the gcode spec.

domenic-d, 2017-05-03:
@tbfleming Thx Todd.

jorgerobles, 2017-05-04:
Refers to #271

tbfleming, 2017-05-04:
Not related to 271

qharley, 2017-08-03:
Even though I know this is temporary it still makes me do a double take when the preview is messed up due to G3 G4 not being implemented.

Is there a permanent gcode parser in the works?

qharley, 2018-06-29:
Still not working?

Been more than a year now.  Can't believe I am the only person bothered by this.
O well...

a7hybnj2, 2018-08-14:
I just ran into this issue as well. I had G code that consisted of just G2 moves. When I loaded it to laser web it looked like nothing was going to happen. The on screen rendering was just blank. When I hit run it did indeed send everything correctly.