LaserWeb/LaserWeb4 #298: Consider moving back to ThreeJS

openhardwarecoza, 2017-05-04:
Doing so should allow progress on:

#25 - rotary axis. Håkan Båstedt already added rotary to lw3s 3D viewer. Including a cylinder gridhelper. Cam was just missing in lw3. Which we now have. Add back threejs viewer and viola! 

#110 time remaining: our old gcode parser did support time calculations 

#147 - we already had TranformControlsjs in Threejs. It just needed a little pretification (at least for vectors. Theeejs has great helper functionality for rotations) 

#187 - existing code available in a private repo of mine for automatic bounding box of actual object - > iso/top/front views zoomed into object fov

#199 - existing code from me and @raykholo s work on cp/reprap workspace.  @lautr3k also more code available with threejs already

#222 - webcam filters - should be easier than raw gl

#248 #297 - the lw3 parser is neae perfect - 3-4 years of work was thrown away

jorgerobles, 2017-05-04:
I did some work, currently unused but ready for review about #110 on https://github.com/LaserWeb/lw.guesstimate

tbfleming, 2017-05-04:
> #110 time remaining: our old gcode parser did support time calculations

The temp parser also does time calculations, but only the bottom slider uses it

openhardwarecoza, 2017-05-04:
@tbfleming , so how do we tap into "time remaining" while running a job? (;

tbfleming, 2017-05-04:
@openhardwarecoza Oops. I remembered wrong. https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/draw-commands/GcodePreview.js computes it and stores it in the vertex arrays. The vertex format changes as new capabilities are added, so probably can't rely on it.

raykholo, 2017-05-04:
> draw-commands/GcodePreview.js computes it and stores it in the vertex

Pardon the oversimplification, but is it possible to keep your low level
webgl/ regl type stuff (if I understand correctly the 3d viewer and live
job rendering are included in this?) and use ThreeJS for doing the higher
level new stuff on the roadmap like rotary and STL?

It's like - can we use the advanced THREE libraries and functionality and
then display it in the non-THREE viewer being used now?  This keeps the
existing stuff fast, which I understand was the reason for going to regl in
the first place?

tbfleming, 2017-05-04:
> can we use the advanced THREE libraries and functionality and then display it in the non-THREE viewer being used now

Definitely worth considering. It's hard to predict how workable this will be. e.g. difference: THREE and the current viewer use different vertex array layouts. THREE uses parallel arrays and ours uses an array-of-struct style (a CPU cache optimization that speeds up the initial creation).

cprezzi, 2017-05-10:
Would it be enough to calculate the time_remaining = (total_lines - executed_lines) / time_running? At least as a quick solution?

closed 2017-07-08 by openhardwarecoza