LaserWeb/LaserWeb4 #300: Postprocessor Engine

jorgerobles, 2017-05-09:
I've been looking for a postprocessor spec to not to reinvent the wheel but have not found anything merely structured.

The simplest engine could be done using a web-worker that receives all the core gcode.
Is up to the post contributor to code whatever he wants. Should be posted on a centralised repository like lw.machines to be approved as a PR. LW could expose those file names depicted on a JSON manifest file. The most LW can actually expose is some comments with a predefined format to hint the postprocessor what's going on core process. I would appreciate concrete implementation ideas.

jorgerobles, 2017-05-09:
Cool thing to implement http://donektools.com/free-cnc-router-software/links-2/

arthurwolf, 2017-05-09:
> Cool thing to implement http://donektools.com/free-cnc
> -router-software/links-2/

You'd have to check their licensing ( or ask permission ), but : Fusion360
( very popular in the "maker" movement lately ) comes with a full set of
really good post-processors, in a js-like format if I remember correctly.
Re-using that format would give you 50+ post-procs instantly, and free
updates just by syncing with fusion

jorgerobles, 2017-05-09:
Good idea. I will ask, but I'm not sure Big A lend us anything :smile:

FabCreator, 2017-05-09:
You could be surprised, the Fusion 360 team are very friendly and keen to keep their project linked with makers! It integrated a post for smoothie, have not used it for a while though i guess it still works?

jorgerobles, 2017-05-09:
https://forums.autodesk.com/t5/computer-aided-machining-cam/post-processor-spec-for-open-source-compatibiliy/td-p/7071215

arthurwolf, 2017-05-09:
Yes, the smoothie post-proc works really great, and we are even working
with them on direct-from-fusion-upload-and-play for smoothieboard ( using
their neat plugin system, another thing you might want to implement later
on, like octoprint did )
that system by the way could be used to make a plugin for fusion that
directly exports vector files to laserweb
workflow is everything

openhardwarecoza, 2017-05-09:
Keen to assimilate makers... Not help. (; someday they'll catch you all
once you are trained and in habit (;

jorgerobles, 2017-05-09:
Of course a company always have a profit, or not to be a company, but I
think is not the makers the ones they want.
The want to hook makers so in case they work for a company, and using that
education base, could attract that company to their side.
Maybe I'm wrong :P

openhardwarecoza, 2017-05-09:
> issuecomment-300093491

Exactly.. Same reason Microsoft gives office365 for free to schools... When
the kids start working they insist to use what they know already

FabCreator, 2017-05-09:
haha @openhardwarecoza  your so untrusting :P 

@jorgerobles you hit the nail on the head, you use fusion 360 now as a student or maker when you start working professionally what you going to use? and start paying for ;) Why you think all autodesk products are free to use for students. (did you know you dont even need to be a student self teaching counts too! or last time i checked it did)

I have no problem with model I have access to a great piece of software and tech support for free now. Start-up licence, when we can afford it will not have a problem paying for it :)

cojarbi, 2017-05-09:
Apple started the same way, not free but giving steep discounts for education ( before was way better than today )

⁣Ariel Yahni​

jorgerobles, 2017-05-19:
https://forums.autodesk.com/t5/computer-aided-machining-cam/post-processor-spec-for-open-source-compatibiliy/m-p/7072212#M29664

--- waiting for instructions - bip - bip - bip ---

jorgerobles, 2017-05-25:
Ouch. I missed Answer Day https://forums.autodesk.com/t5/autodeskhelp/autodesk-answer-day-may-24th-2017/ba-p/7009811

Last notice I've got on this by Autodesk were weeks ago. I don't even know if will get a short term answer. Any ideas?

tbfleming, 2017-06-11:
We probably need to nail down the use case: what will users need custom post-processors for? Will they solve anything that adding new config settings (e.g. #334) won't solve?

jorgerobles, 2017-06-11:
Probably not. Including flexible settings, to Mach3 and Claudio's support on Marlin Kimbra could complete the market :D

jorgerobles, 2017-06-11:
Maybe I'm wrong but I think is not just what postprocessors could do (as adding support to drag knife, plasma cutters with third party integration), but kind of software architecture. I fear that current one could reach top of maintenance, new features bloating configs and corner cases. I'm thinking of an event based (just like redux but for interfacing and aspectual programming) that could enable the things a user needs and disabling what not. 
What do you think? (I'm sure you have some deep knowledge on this area :) )

tbfleming, 2017-06-11:
Something we'd probably want to avoid: post processors which take gcode in. That's a very difficult road to go down; look at Skeinforge. It produces gcode near the beginning of its pipeline then all stages after that take gcode in. Stages insert comments to communicate with later stages.

jorgerobles, 2017-06-11:
BTW. Has no answer yet by [big A](https://forums.autodesk.com/t5/computer-aided-machining-cam/post-processor-spec-for-open-source-compatibiliy/m-p/7072212#M29664) about this topic. Sure they are busy.

jorgerobles, 2017-06-11:
> Something we'd probably want to avoid: post processors which take gcode in. That's a very difficult road to go down; look at Skeinforge. It produces gcode near the beginning of its pipeline then all stages after that take gcode in. Stages insert comments to communicate with later stages.

Yes. But then We need to develop intermediate language to finally convert to Gcode, like Fusion360 does. Isn't it?

tbfleming, 2017-06-11:
Not a language, a Javascript interface. The downside is it will slow down most users who don't need a postprocessor.