LaserWeb/LaserWeb4 #302: LW4 does not render filleted edge appropriately?

PeterGrace, 2017-05-10:
[LW4-ExtruderPlates.json.txt](https://github.com/LaserWeb/LaserWeb4/files/990846/LW4-ExtruderPlates.json.txt)
[ExtruderPlate.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/990847/ExtruderPlate.dxf.txt)

Attached is dxf and workspace file.

I created this dxf design in QCad, saved it in R12 format.  When I imported it, I noticed that my edges that I rounded with the "Round" tool in QCad (to produce a fillet) does not register properly in the LW4 workspace, it instead looks like a diagonal line.  

Thinking that it might be an issue with individual rays not being a polyline, I instead created a polyline of the sum of the rays and it still shows diagonal in the LW4 workspace, even though QCad shows an appropriately rounded fillet.

openhardwarecoza, 2017-05-10:
Explode the curve to polyline in Qcad (x, p)

PeterGrace, 2017-05-10:
@openhardwarecoza  I can confirm that does enable the part to render correctly.  Is the symptom I'm seeing a problem with LW4 dxf rendering, or is QCad breaking another R12 rule?

tbfleming, 2017-05-10:
It's a nightmare to try to support everything DXF supports, so we only support a subset.

openhardwarecoza, 2017-05-10:
Qcad... Lw just wants polylines. If you dont convert curves to polylines...
You arent giving us polylines

openhardwarecoza, 2017-05-10:
Read
http://cadsetterout.com/autocad-tutorials/cad-to-cam-cleaning-up-dwg-files-for-cnc
for some generic advice

closed 2017-05-10 by openhardwarecoza