LaserWeb/LaserWeb4 #303: Moving onto production builds

jorgerobles, 2017-05-12:
I've been wondering of compiling production builds instead development ones https://facebook.github.io/react/docs/optimizing-performance.html#use-the-production-build

We have an steady build now with no critical bugs, we could do some benchmark on this side.

openhardwarecoza, 2017-05-12:
> performance.html#use-the-production-build

Theres a bug at the moment somewhere in lw4 that causes npm run bundle-prod
to fail

jorgerobles, 2017-05-12:
then put to work! :rofl:

jorgerobles, 2017-05-12:
I've solved bundle-prod on (bundle-prod own branch) https://github.com/LaserWeb/LaserWeb4/commit/7e670004c3e13ad558fdf8d7189d5ff27c2dde00

I haven't seen any side effects, but maybe are. Please take a look.

Also, should disable react middleware if on prod. any ideas?