LaserWeb/LaserWeb4 #305: [Alert] Job Freezing

jorgerobles, 2017-05-14:
Hi all,

I've been experimenting continuous random job freezing. I suspect have to do with latest changes, maybe https://github.com/LaserWeb/LaserWeb4/commit/0def27607b08554183cedea0e3e7ce707f401265 commit. Or gcode generation (less probable)
Will revert ASAP. 
Be aware if you are testing, please report. Thanks

jorgerobles, 2017-05-14:
@tbfleming, @cprezzi, @openhardwarecoza

closed 2017-05-17 by jorgerobles

tbfleming, 2017-05-17:
Was it 0def276?

jorgerobles, 2017-05-17:
I haven't tried, but is reverted so if it happens to anyone, let me know.
I suspect boxes.py svg is not quite right, also.

jorgerobles, 2017-05-20:
Have experienced segment clog. But weird thing is that the first time, same
settings and gcode did not happen. Tried again same work happened... Have
to test grbl to compare.