LaserWeb/LaserWeb4 #308: Global operation starting height

domenic-d, 2017-05-16:
Setting the starting height for individual operations is useful, but we find ourselves and customers using one material height at a time, most of the time.

With many operations being used at a time it becomes cumbersome to systematically go through each one when a different material thickness is being used.

A global starting height would in actual fact prove more commonly used.

Without wanting to add extra options, I just wanted your thoughts on this and whether you agree with this and/or have any suggestions on how to work this in?

tbfleming, 2017-05-16:
I suspect this may apply to multiple fields. Maybe a button by each field which copies the value to all other operations?

domenic-d, 2017-05-16:
I see, this feature could prove useful with the absence of a 'duplicate' operation.

cprezzi, 2017-05-16:
What if we just remember the last used value and use it for every new operation?

jorgerobles, 2017-05-16:
I'm already working on the duplicate button. :)

cprezzi, 2017-05-16:
I would prefer the "last value" way, as this would also speed up the process of designing multiple different operations, not only duplicates.

jorgerobles, 2017-05-16:
Ok, will think about the better way to implement.

openhardwarecoza, 2017-05-16:
I also prefer the "last value" and sometimes even useful if that can persist in localstorage between sessions and between workspace resets.  I find myself buying a sheet of some material, using it for everything, lol. Maybe thats my budget's problem not lw's lol

jorgerobles, 2017-05-16:
Ok, it's easy to save last operation selected on localstorage, but are you sure that's the way?  a redux key currentoperation is stored and may be used as reference-- that is the same as duplicate. 

It's unclear for me the way it has to proceed.

BTW- the start height is a setting already, that fills every operation that has is empty...

cprezzi, 2017-05-16:
I wouldn't store the whole operation, just all the separate field values (independant of operation type).

jorgerobles, 2017-05-16:
Ok then. We are then talking specific, hardcoded, fields, `start_height` and `cutrate`?

cprezzi, 2017-05-16:
That was my idea. All fields (that do not have a differently defined init value) would just use the last used value of that field. Applying something from the material database would override the fields value, but not the saver "last value".

domenic-d, 2017-05-16:
Repeating the previous value for a new operation only partially addresses things.
From my initial post, the time consuming part is when you need to go back and use the same set of operations for a different thickness of material. This is the time consuming part.

In actual fact, I would almost say it's not appropriate to have previous operation values automatically filled when creating a new one as it could lead to users not checking the values that are blank and obviously important.

jorgerobles, 2017-05-17:
@domenic-d then, an "link all button" on start height? The button could will be located on the start height field each operation next its name or beign a context button over the field name. if toggled any change on start height on that field will automatically spread that value to all operations.

jorgerobles, 2017-05-17:
Ok, I've implemented a contextual menu available on operation fields `startHeight`, `cutRate`, with the command _Spread to all Ops._ This copy the value of the actual entered field to all the other operations.
It's generic so it's base to extend any field with some context menu.

jorgerobles, 2017-05-18:
@domenic-d can you take a look to the latest code? Is that context menu closer to your idea?

domenic-d, 2017-05-18:
@jorgerobles Clever solution without cluttering up the UI with more buttons.
I think it should be probably called 'Copy to all Ops' instead of 'Spread....'

From my testing it seems to be working correctly.

Thank you.

jorgerobles, 2017-05-18:
Copy then :)

closed 2017-05-18 by jorgerobles