LaserWeb/LaserWeb4 #309: No supported firmware detected. Closing port COM3

Chapster5, 2017-05-16:
Hi all,

Just flashed GRBL 1.1F successfully to my Arduino Nano 328P however when I click connect in the Laserweb interface I keep getting this error:

Connecting Machine @ USB,COM3,57600baud
 Machine connected
 No supported firmware detected. Closing port COM3
 Machine disconnected

I have flashed using both XLoader, which said successful and I tried a manual compile and upload through Arduino IDE again that said it was successful I've restarted laser web a few times ive plugged and unplugged the machine both power and USB and still it can't connect. Any ideas?

Cheers

closed 2017-05-16 by Chapster5

cprezzi, 2017-05-16:
Default grbl baud rate is 115200. Config LW4 accordingly.

mkeyno, 2017-05-23:
@Chapster5  how to you solve your problem? @cprezzi both part have same baud rate of 115200  but not success 
```
 Server error: Opening COM11: Unknown error code 121
 Machine disconnected
 Connecting Machine @ USB,COM11,115200baud
 Machine connected
 No supported firmware detected. Closing port COM11
 Machine disconnected
```

cprezzi, 2017-05-24:
@mkeyno Your problem is different. This here was just the wrong baudrate.