LaserWeb/LaserWeb4 #312: Control of grid size and resolution

domenic-d, 2017-05-18:
The ability to set the size of grid and spacing.

jorgerobles, 2017-05-18:
Size of grid is on settings/application :)
I think we could move all grid settings to a tab @tbfleming ?

domenic-d, 2017-05-18:
The current size setting is good. Only missing the ability to set the grid spacing.
Also, the negative grid being there looks great but may confuse non-technical users.

cprezzi, 2017-05-18:
I don't think the grid will confuse users, especially with the new visible machinespace and origin offset. That makes it very obvouis for every user.

jorgerobles, 2017-05-19:
Added minor/major grid settings:application. Use carefully it could affect display performance.
Negative coords are ok for me too :)

closed 2017-05-19 by jorgerobles

flynnjs, 2018-06-04:
Any way to change the grid so that -ve goes up and +ve goes down? We have a unit with the origin at the top left and if we use all positive coordinates it cuts upside down. We did reverse Y axis in cutter firmware and set origin to Y max but not sure that a good way to operate it.

cprezzi, 2018-06-06:
Positive directions should always be in right, back and up direction, independant of the machine origin! If your origin is top left, then your work space is in negative y range.
It is the correct solution to configure the firmware (or swap the motor wires) to get the correct axes direction.