LaserWeb/LaserWeb4 #315: Gcode freeze on 4.0.732 / 4.0.78

cojarbi, 2017-05-20:
Gcode generation freeze under the following:

Load image file, create raster operation, press generate:
- In most cases generation will freeze at random %. Pressing stop does nothing. Restart is needed
- If it does generate, pressing generate again will freeze the process, stop does not clear. Restart needed
The above mostly happens to me with random png's, jpeg and jpg are a hit or miss

Other cases
- Pressing generate with no files or operations will freezes the process, stop does not clear. Restart needed.
- Disable and operation (Enable/Disable button) also freezes the process, stop does not clear. Restart needed.

jorgerobles, 2017-05-20:
Will check :)

cojarbi, 2017-05-20:
[Laserweb-Workspace.zip](https://github.com/LaserWeb/LaserWeb4/files/1016321/Laserweb-Workspace.zip)

jorgerobles, 2017-05-20:
Thanks. I've reverted the responsive progressbar. Should work now, @cojarbi cannot compile until monday :(

@tbfleming Trying to throttle rendering, as progressbar is choking at high concurrency, found
LW processing is almost instant (x2~4 on vector and x10 on raster) if logger is disabled.

cojarbi, 2017-05-21:
@jorgerobles compiled and tested, all seems good

closed 2017-05-23 by jorgerobles