LaserWeb/LaserWeb4 #317: Jogging with cursors

garciasa, 2017-05-21:
Hello guys,
is there any chance to include checkbox in the ui or wherever for indicating to LaserWeb we can jog with just  cursors?? Now it´s working if you press alt+up, alt+down and so on, but I´m using a standalone numeric keyboard, and alt key is not there, and my machine is not close to my computer...

At the moment I did this small fix, but I don´t know, if we also use just up, down and so on, it will affect to other part of the software.

`    @keydown(['alt+right', 'right'])
    jogRight(event) {
        event.preventDefault();
        this.jog('X', '+')
    }

    @keydown(['alt+left', 'left'])
    jogLeft(event) {
        event.preventDefault();
        this.jog('X', '-')
    }

    @keydown(['alt+up', 'up'])
    jogUp(event) {
        event.preventDefault();
        this.jog('Y', '+')
    }

    @keydown(['alt+down', 'down'])
    jogDown(event) {
        event.preventDefault();
        this.jog('Y', '-')
    }`

Cheers,

jorgerobles, 2017-05-21:
Well, as Jog is scoped to component, I'm not sure what could go wrong, maybe trying to type an override. But just for safety, we can change your arrow codes to keypad num codes (96 to 105) if keypad setting (ported from LW3 is enabled). What do you think?

garciasa, 2017-05-22:
Hi Jorge,
that sounds good to me.... I will have a look with keypad enabled and those codes, and I will get back to you, but that should be enough, at least for people who is working with a poor keypad as pendant  like me hahaha ...

Thanks.

jorgerobles, 2017-05-22:
Don't worry, I'll do for you. No fuzz.

jorgerobles, 2017-05-23:
Done!

closed 2017-05-23 by jorgerobles