LaserWeb/LaserWeb4 #318: LW UI Performance hit

jorgerobles, 2017-05-21:
I'm having hard time with parallel webworkers concurrency and react response. 
The most patent case is that LW is unresponsive on Gcode generation due to parallel processing choke.
Can anyone lend me a hand?

tbfleming, 2017-05-21:
Try limiting the number of workers to the number of CPU threads - 1. Ex: my desktop is a 4-core 8-thread machine, so it'd be 7 on it. There is no browser API that reports this so it'd have to be a setting.

jorgerobles, 2017-05-21:
Is a setting indeed :) I've done a naive approach using setinterval to
percent dispatching. Seems good but maybe there's a better way.

jorgerobles, 2017-05-21:
I'm quite in shock. A ~44sec computing to merely 1s540

closed 2017-05-23 by jorgerobles