LaserWeb/LaserWeb4 #321: toggleable Disable onChange evaluation of textboxes?

PeterGrace, 2017-05-23:
I'm currently working with a very large (~30MB) dxf with lots of paths.  When I added an operation on those paths, and then attempted to edit the "Name" field of the path, it took the program over a minute to fully recognize I typed the letters "engraves" -- the letters showed up one or two at a time with long periods of "thinking."

I'm assuming that this happens because some event handler is catching onChange on all of the textboxes in an operation declaration and is causing some sort of iteration of all paths.  I'm wondering if this was something done intentionally, or whomever wrote the code did not realize it would struggle with large sets of paths.

What I'm suggesting is that if this functionality is necessary, could we have a toggle that enables/disables the processing and instead gives a button to trigger that processing after I've completed all of the necessary changes to the operation (e.g. beam diameter, laser power, cut rate, start height, etc).  Having this would speed up my cut setup time.

I'd dig in and have a look myself but I'm not very competent with node.

cprezzi, 2017-05-23:
Could you please upload your file so we can replicate your problem?

jorgerobles, 2017-05-23:
@tbfleming I think that is reaching redux update capability. Could go for updating onBlur instead onChange?

tbfleming, 2017-05-23:
@jorgerobles let's see what's happening first. A slow onChange will mean a slow onBlur. Less frequent but will still be annoying.

PeterGrace, 2017-05-23:
[LW4-50mm-back.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1022671/LW4-50mm-back.json.txt)

I've attached the workspace here.  At this point I'm wondering if I should just do this as a raster instead of vector path cuts.

jorgerobles, 2017-05-23:
@tbfleming It's rather heavy file, but runs decently on my desktop. The clog seems located on workspace selected vector animations, rather than textboxes.
@PeterGrace deselect any vector prior to change textfields, and report plz. Thanks

jorgerobles, 2017-05-23:
@PeterGrace ---- 
>At this point I'm wondering if I should just do this as a raster instead of vector path cuts.

Yes, I do that way, and is ok :)

jorgerobles, 2017-05-25:
Going to #216 then. Closing.

closed 2017-05-25 by jorgerobles