LaserWeb/LaserWeb4 #322: No supported firmware detected. Closing port

mkeyno, 2017-05-24:
I run laserweb4 rev 4.0.728 with GRBL 1.1f how ever same as [this  issue ](https://github.com/LaserWeb/LaserWeb4/issues/309) ,it disconnected to the machine , both part set to 115200baud and my board simply connected to other CAM software 

```
 Server error: Opening COM11: Unknown error code 121
 Machine disconnected
 Connecting Machine @ USB,COM11,115200baud
 Machine connected
 No supported firmware detected. Closing port COM11
 Machine disconnected
```

jorgerobles, 2017-05-24:
@cprezzi https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/333 ??

cprezzi, 2017-05-24:
@mkeyno We need the boards welcome string for compatibillity check and selecting the correct firmware protocol. If this string is not received within 2s, we disconnect.

Please try this:
- Open the Arduino IDE
- Seletct the same port you used with LW4
- Open the serial monitor
- You should see the welcome string *Grbl 1.1f ['$' for help]* after 1-2s (without sending something!)

cprezzi, 2017-05-24:
If you also don't get the welcome message with the serial monitor, then you are either on the wrong port or there is some other problem. You could then try to change the COM port number to something lower (1-6) via Windows Device Manager.

mkeyno, 2017-05-24:
thanks @cprezzi , but as mentioned its already work with other CAM software terminal and my GRBL board correctly send the `Grbl 1.1f ['$' for help]` , maybe it should extend the time or in this case send reset command `0x8` to be sure capture the welcome message , however I might add I use the Bluetooth serial bridge and it may probably take longer time regard to wire connection

cprezzi, 2017-05-24:
It doesn't matter if it works with other gcode sender or CAM as they probably don't rely on the welcome message. The test with the Arduino IDE would help me to analyze the problem.

What exactly is this bluetooth serial bridge? 
The Arduino only sends the welcome message, if it is connected via USB, not via serial pins (rx/tx)! 

It's also possible that it just takes a bit too long until we get the message. In this case, I could probably make the timeout configurable via an environment variable.

mkeyno, 2017-05-24:
thanks @cprezzi  , yes it work on Arduino IDE  as well , and BT serial bridge is pretty normal today's, I use HC-06 to to control my cnc machine  or other wireless serial communication projects , it's just need to connect rxtx pins , however as you mentioned , environment parameters should changed according to extend timeout , can you tell me where the proper variable is ?

openhardwarecoza, 2017-05-24:
Bt bridge doesnt autoreset like usb does. Hit reset button on arduino and connect in lw at same time. Grbl spits welcome string out when usb port opens, pulling dtr low, resetting arduino. Bt doesnt do that

cprezzi, 2017-05-24:
@mkeyno You realy MUST be EXACT if you like to get support. 
Do you realy get the welcome message when connecting with the serial monitor of the Arduino IDE WITHOUT sending anything? I don't think so.

mkeyno, 2017-05-24:
@cprezzi yes , it is show the welcoming message as it expected but as the @openhardwarecoza  truly mentioned its due to lack of autoreset like usb and its handshaking  longer than wire serial , and I think if the waiting time extended its easily solve the problem , but this version is in binary  and I don't know how to change the time

cprezzi, 2017-05-26:
The next binary will read the FIRMWARE_WAIT_TIME from an environment variable so you can place a .env file to the installation folder with `FIRMWARE_WAIT_TIME=10` to extend the timeout to 10s.

cprezzi, 2017-05-26:
Did you know that you can also reset the board by pressing `CTRL-x` in LaserWeb?

mkeyno, 2017-05-26:
thanks  very much @cprezzi , yes I know that , however it must first connect to COM at first , also I find .env file in my laserweb3 installation folder  but not such things in laserweb4 , I hope it will be ready soon

cprezzi, 2017-05-26:
It will be comming with the next binary version. There will be no .env file created automatically. You have to do create a text-file in the LW4 installation folder with the line `FIRMWARE_WAIT_TIME=10` and name it .env.

funinthefalls, 2017-05-27:
That is a great feature @cprezzi Any other items going to be user settable in that file?

mkeyno, 2017-05-27:
I agree if this variable extended , such as  engraving setting for the specific firmware , and special setting for setup the firmware , for example special tab page display the command,variables and description need to setup specific firmware

mkeyno, 2017-06-04:
hi @cprezzi , is v4.0.733-79 version has the mentioned feature ?

cprezzi, 2017-06-04:
Yes, ...-79 has the FIRMWARE_WAIT_TIME feature.

cprezzi, 2017-06-04:
@funinthefalls All available params of the .env file arre:
- WEB_PORT={IP port of server}
- VERBOSE_LEVEL={0..3}
- LOG_LEVEL={0..3}
- DRO_DECIMALS={0..4}
- FIRMWARE_WAIT_TIME={0..x}

mkeyno, 2017-06-05:
@cprezzi its connected only for 10 seconds (as it set)  and throw the issue 141  , I think the problem lay somewhere else because its work when it conned by serial connection 
log
```
Connecting Machine @ USB,COM19,115200baud
 Machine connected
 No supported firmware detected. Closing port COM19
 Machine disconnected
 Connecting Machine @ USB,COM19,115200baud
 Machine connected
 No supported firmware detected. Closing port COM19
 Machine disconnected
 Connecting Machine @ USB,COM19,115200baud
 Machine connected
 
 Firmware grbl 1.1f detected
 [MSG:'$H'|'$X' to unlock]
 Work Offset: 0.00 / 0.00 / 0.00 / NaN
 Work Offset: 269.94 / 59.13 / 6.00 / NaN
 Work Offset: 0.00 / 0.00 / 0.00 / NaN
```

cprezzi, 2017-06-05:
@mkeyno Disabling the timeout would not help as the server needs the welcome string to select the right firmware protocoll.
You only get this problem, because you use the bluetooth bridge. You should consider to connect directly by USB (or live with manual reset).

mkeyno, 2017-06-08:
@cprezzi  can you add another optional parameter such as COMMAND_ON_CONNECT so it could send the reset command to get the welcome message ?

cprezzi, 2017-06-08:
@mkeyno Ok, I have added a new environment variable 'RESET_ON_CONNECT'. The server sends a ctrl-x (reset) command to the board, if this variable is set to 1.

This feature will be available with backend version 4.0.96.

mkeyno, 2017-06-08:
you are my man , thanks @cprezzi

mkeyno, 2017-06-08:
@cprezzi I'm curious about how & where to add these env parameters in source , actually I intend to test some specific parameters , can you give me some clue how to rebuilt the source and create the binary program

cprezzi, 2017-06-08:
@mkeyno Uhh, that is a lot to explain! It's a whole chain of requirements you have to fulfill (git, node.js with npm, multiple git repositories...).
If you are an experienced developer, start by reading https://github.com/LaserWeb/LaserWeb4/wiki/How-to-Contribute. After you have forked and cloned LaserWeb4 to your local machine, you need to run 'npm run installdev' to get all the dependencies installed. You also need to fork and clone the lw.comm-server repo to add/change something on the server side (like the env variables). The process to compile the electron installer is basically explained in the electron_bundler branch of lw.comm-server.

cprezzi, 2017-06-08:
The whole server code is in the server.js file of the lw.comm-server repo. config.js is where compile params and environment variables are defined.

cprezzi, 2017-06-08:
But the environment variables are no long-term solution. They will probably be replaced in some time with machine settings in the frontend.

mkeyno, 2017-06-08:
@cprezzi  actually I did and tried to install it on my machine but I'd not be sure whether this ticket be suitable for such matter , caz it seems I miss couple of libs which is not fully addressed in the https://github.com/LaserWeb/LaserWeb4/wiki/How-to-Contribute page, kindly let me know if there is special ticket for that

gerwinbrunner, 2017-07-03:
I'm experiencing the same problem with Laserweb 4.0.735 and Grbl 1.1f. Running on OSX 10.12.5.
Any new infos about the issue?

I tried the .env file, but it seems not to work the same way for OSX (I made a file LaserWeb.env that lives in the same folder as the app)

gerwinbrunner, 2017-07-03:
Here is the console output
[image]

gerwinbrunner, 2017-07-03:
Some more observations:

I connected with the Serial App directly and the string `Grbl 1.1f ['$' for help]` shows up within around 3 seconds.

I also tried to flash and downgrade to Grbl 1.1e  -  same result.

gerwinbrunner, 2017-07-03:
So I just checked out the source code and started it locally, set the LOG_LEVEL=3,  the VERBOSE_LEVEL=3 and the FIRMWARE_WAIT_TIME=20.

Still `No supported firmware detected`

Here is the complete output:
```
webpack: Compiled successfully.
App connected! (id=0)
INFO: Requesting Server Config 
INFO: Connecting to USB,/dev/cu.Repleo-CH341-00001014,115200
INFO: Connected to /dev/cu.Repleo-CH341-00001014 at 115200
Sent: version
Recv: error:2
Sent: {fb:n}
Recv: error:1
Sent: M115
Recv: error:20
No supported firmware detected. Closing port /dev/cu.Repleo-CH341-00001014
INFO: Port closed
```

Any ideas?

gerwinbrunner, 2017-07-03:
OK... traced it down... 
The setting that makes it work is just
`GRBL_WAIT_TIME=10`
Then it works like a charm :)

Maybe the initial value for this should be >0.5s

gerwinbrunner, 2017-07-03:
So tested it with the release versions:

`v4.0.735-96` does not yet seem to support `GRBL_WAIT_TIME`
`v4.0.735-109` does not have a OSX dmg - so can't test :(

So I guess there is no release version for OSX that works right now.

cprezzi, 2017-07-03:
Thanks for reporting! 
The env file has to be named ".env" (without any filename in front of the point).
The GRBL_WAIT_TIME environment variable is already available in backend version 109.

gerwinbrunner, 2017-07-04:
@cprezzi thanks for pointing out the correct filename.
I tried it, but it seems that it is not working on an OSX app.

Here is what I did:
1) I placed a '.env' file in the same folder as LaserWeb.app
[image]

2.) I opened the LaserWeb.app package (right click on the app -> 'Show Package Contents')
and placed the the .env file in all of the following folders:
- /
- /Contents
- /Contents/MacOS
- /Contents/Frameworks
- /Contents/Resources

It worked in none of those.

And here is the content of the .env file
```
FIRMWARE_WAIT_TIME=30
GRBL_WAIT_TIME=10
LOG_LEVEL=3
VERBOSE_LEVEL=3
```

I guess the issue is that the path cannot be resolved by 'dotenv' that is used by lw.comm-server
So I wasn't really able to dig deep enough to figure out which path it actually looks for, but none of the ones I tried worked.

Any ideas?

cprezzi, 2017-07-04:
Hm. It works just fine on Windows Electron App. So must be something Mac related. Unfortunately I'm not a Mac guy. I'll ask our Mac dev to test...

gerwinbrunner, 2017-07-04:
@cprezzi thx
How are the electron apps actually built? 
Can you point me to some info about it. I only found the old LaserWeb3 (deprecated) repo for electron.

cprezzi, 2017-07-04:
The electron app is built from https://github.com/LaserWeb/lw.comm-server/tree/electron_bundler

openhardwarecoza, 2017-07-04:
https://github.com/LaserWeb/lw.comm-server/blob/electron_bundler/README-ELECTRON.md detailed instructions

openhardwarecoza, 2017-07-04:
Seeing as its a CH340 based device, you did remember to install the driver? https://tzapu.com/making-ch340-ch341-serial-adapters-work-under-el-capitan-os-x/ for example (google Mac OSX <your version> CH340 driver)

gerwinbrunner, 2017-07-04:
Jup, driver is installed (and works with the checked out dev version and the GRBL_WAIT_TIME=10)

jsmillie7, 2017-08-15:
Hi, has there been any solution regarding this for MacOS yet?

cprezzi, 2017-08-15:
@jsmillie7 Not yet, but I'm working on integrating the "Reset on connect" option to the frontend settings.

jorgerobles, 2017-10-21:
What's the status of this issue? still alive?

jyauxi, 2017-12-05:
For macOS, setting the environment variable needs to be done on the info.plist file.

To do this,

- right click on the `LaserWeb.app`
- "Show Package Contents"
- Navigate to Contents
- Edit info.plist with the following:

Add the following within `<dict>    ~~~~   </dict>`
```
<key>LSEnvironment</key>
	<dict>
		<key>GRBL_WAIT_TIME</key>
		<string>10</string>
		<key>LOG_LEVEL</key>
		<string>3</string>
		<key>VERBOSE_LEVEL</key>
		<string>3</string>
	</dict>
```

After editing the info.plist file, I can get LaserWeb to detect the GRBL controller.

Protoneer, 2017-12-19:
Is there a way to specify in the config that LW will be running with GRBL? (Instead trying to detect it automatically)

cprezzi, 2017-12-19:
That's not what we want. Grbl detection is very stable with USB. Problems just can be caused from WLAN or Bluetooth bridges, that do not support board reset. Those rare problems can be solved with the existing env variables.

Protoneer, 2017-12-19:
Currently my RPI CNC hat does not work with LW because it uses the hardware
serial. The hat has reset circuitry via some of the other RPI gpio pins.

Reset via serial works from bCNC and CNCJS.

I am creating a RPI preconfigured images and can modify LW code but I would
prefer to stick to the standard build.

Protoneer, 2017-12-19:
I just noticed RESET_ON_CONNECT in the GRBL LPC wiki. I will try that first.

cprezzi, 2017-12-20:
RESET_ON_CONNECT is the correct var for that. If enabled, it sends a reset command after connecting (for serial rx/tx connected devices).

jsmillie7, 2017-12-21:
Hey all, not sure if this is helpful, but after adding jyauxi's code to the info.plist, it gives me enough time to manually reset the arduino, which then connects to the bluetooth connection. A small workaround for MacOS devices.

gdenicolas, 2017-12-22:
I am running Laserweb 4.0.990, Board: spark concepts xpro v3, windows 10.

I get the same error messsage. When I connect through Universal Gcode Sender, I get connected, but not with Laserweb. Pictures attached
[image]
[image]
[image]

cojarbi, 2017-12-22:
I have the same board on 2 CNC machines with no issues. 

Could you without knowing be keeping the port open in UGS?

⁣
Ariel Yahni​

gdenicolas, 2017-12-22:
@cojarbi thanks. I close the connection, and close UGS before opening Laserweb

cprezzi, 2017-12-23:
@gdenicolas Is your board connected with a normal USB cable?
You could try to increase the FIRMWARE_WAIT_TIME and GRBL_WAIT_TIME in the ".env" file (see above).

gdenicolas, 2017-12-23:
@cprezzi I am using an USB cable that was working two weeks ago, and I tried with another..
I tried the parameters in the .env  file.
When I connect through UGS, the response from the board is immediate, that's why I think that this parameters are not the answer.
Thank you for your ideas and support

Setialife, 2019-02-18:
@cprezzi i have this issue too, this is my config , do i must to add .env ? i use esp32s & esp8266 esplink,, thanks before

require('dotenv').load({ silent: true });

var config = {};

config.webPort = process.env.WEB_PORT || 8000;
config.serverVersion = '4.0.127';
config.apiVersion = '4.0.6';

config.verboseLevel = process.env.VERBOSE_LEVEL || 1;
config.logLevel = process.env.LOG_LEVEL || 0;
config.resetOnConnect = process.env.RESET_ON_CONNECT || 1;

config.posDecimals = process.env.DRO_DECIMALS || 2;
config.firmwareWaitTime = process.env.FIRMWARE_WAIT_TIME || 10;
config.grblWaitTime = process.env.GRBL_WAIT_TIME || 10;
config.smoothieWaitTime = process.env.SMOOTHIE_WAIT_TIME || 1;
config.tinygWaitTime = process.env.TINYG_WAIT_TIME || 1;

config.jobOnStart = '';
config.jobOnFinish = '';
config.jobOnAbort = '';

module.exports = config;

[image]

cprezzi, 2019-02-18:
@Setialife Your config.js sets the default for "RESET_ON_CONNECT" to 1, if it's not defined in .env file. Just check that you don't have RESET_ON_CONNECT=0 in your .env file.

Setialife, 2019-02-18:
i installed esp websocket too, and still can't to reset the grbl , this is my env file,, am i wrong ? or i must delete env file ?
[image]
[image]

[17 more replies]