LaserWeb/LaserWeb4 #323: Feature request - joystick for jogging.

funinthefalls, 2017-05-24:
I was hoping that we could get joystick jogging added to the current feature set.

Thanks in advance
Anthony

jorgerobles, 2017-05-24:
@funinthefalls I would rather use a Joystick to Keyboard mapping software. Using Joystick seems a bit complicated to include right now.
Have you tried Live Jogging feature?

funinthefalls, 2017-05-24:
I can give mapping software a try. I had spoken to Todd about adding joystick ability, and he said to go ahead and ask for it. But I understand you guys have your plate full right now, so I will try your suggestion, and maybe revisit this when you guys are not so busy. Thanks for the quick response, I'll close this for now.

jorgerobles, 2017-05-25:
Highly experimental Gamepad support @funinthefalls, @openhardwarecoza 
https://github.com/LaserWeb/LaserWeb4/tree/gamepad

funinthefalls, 2017-05-25:
Thanks. How do I run this to rest it on my system?

jorgerobles, 2017-05-25:
What OS do you use? Win? I could try to compile this branch for you, or maybe could you install development stack (faster for dev/testing)

jorgerobles, 2017-05-25:
If you use linux, here is the app http://laserweb-release.s3-website-us-east-1.amazonaws.com/LaserWeb/LaserWeb4/211/211.1/dist/lw.comm-server-4.0.78-x86_64.AppImage

funinthefalls, 2017-05-25:
I am on Win10. If you can give me a list of what I need, I will certainly install the dev stack.

jorgerobles, 2017-05-25:
@funinthefalls https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/v4.0.732c-78

How to:

- Go to settings -> application. Enable **Use gamepad**
- Connect your machine
- Go to Control, Connect your gamepad. should appear on console as Gamepad connected. 
- Lets play! (left jogger)
- Does not work outside jog mode. 

Caveats:
 
- Currently cannot go diagonal (have to review jog command)

Bonus!
- This release includes delete documents selected pressing "del" if mouse is over the workspace.

funinthefalls, 2017-05-25:
Wow that was quick. Thank you for getting this to me to test.

funinthefalls, 2017-05-26:
Will be going out this weekend to get a new gamepad for testing. Any reccomendations on make or mode?

tbfleming, 2017-05-26:
[image]

jorgerobles, 2017-05-26:
I got a rainbow chinese sh** gamepad and goes fine

funinthefalls, 2017-05-27:
Nice setup Todd. I have actually considered buying a CH full flight sim joystick,pedal,yoke,throttle package for building a Drone control station, as well as for flying my model planes and helis. I almost bought it then was afflicted with the laser and CNC disease, and all the fun money has been going to this new addiction.

Thanks for the info @jorgerobles

funinthefalls, 2017-05-31:
Got a gamepad finally, goiong to test over the next couple of days and get back to you.

funinthefalls, 2017-06-06:
It seems to work OK, but has some jittery movements sometimes. Not sure if it is the cheap gamepad cable, or something in the coding. I am leaning towards it being a crappy cable, as the quality does not appear to be there.

jorgerobles, 2017-06-07:
Ok, will include to next release, so anyone can test :)

funinthefalls, 2017-06-07:
Sounds like a plan. Once again, thank you for adding this feature.

closed 2017-06-08 by jorgerobles