LaserWeb/LaserWeb4 #324: Work offset messages ...

KirAsh4, 2017-05-25:
Running a K40 Laser cutter with a Cohesion3D board. After installing the latest Win64 binary release, I'm now getting a bunch of these messages popping up.
[image]

closed 2017-05-25 by jorgerobles

KirAsh4, 2017-06-01:
Hey Jorge, this is still present in the most recent Windows release. Forgot to update the master branch?

jorgerobles, 2017-06-01:
I got rid of _continous_ work offset messages on that commit  c3e42cc . Now should only appear the first time the machine shows it or changed. Do still appear to you? 

[image]

KirAsh4, 2017-06-01:
Yep, a continuous stream. I'm running 4.0.733-79-x64 on Win10.

jorgerobles, 2017-06-01:
Is it GRBL or Smoothie? @cprezzi  is sthis behaviour firmware dependant?

KirAsh4, 2017-06-01:
I'm running a Cohesion3D board which runs Smoothie firmware.

cprezzi, 2017-06-01:
@jorgerobles Grbl is reporting WCO when changed, but less often with status reports (not every). Smoothie (over USB) is reporting MCO and WCO in every position report. I cache the offsets in lw.comm-server and only send them to the client when something changed. At least that's what it should do.

cprezzi, 2017-06-01:
@KirAsh4 Are you sure you user smoothies *firmware-cnc.bin*?

jorgerobles, 2017-06-01:
@cprezzi that is what I did in the commit. Is weird it still happens. Is working ok on my board :/

cprezzi, 2017-06-01:
@KirAsh4 It would help if we could see the lw.comm-server logs. Could you please create a ".env" file in the LW installation folder with the following text in it: `LOG_LEVEL=3`
This activates writing a logfile.txt in the installation folder.

cprezzi, 2017-06-01:
@jorgerobles I think I found the reason (toggling around +/- zero) and fixed it with lw.comm-server commit https://github.com/LaserWeb/lw.comm-server/commit/c5b3cb496767bad96534986f65792728cfcc1a3f

cprezzi, 2017-06-01:
I have tested and my commit doesn't fix the problem. I am working on it.

cprezzi, 2017-06-01:
Ok. Initializing the local position and offset vars to float 0.00 instead of int 0 seems to help :)
https://github.com/LaserWeb/lw.comm-server/commit/5f73724dd9b46ef7e0043971c72920cdd7cc1bde

KirAsh4, 2017-06-01:
@cprezzi, yep, I'm using firmware-cnc. It sounds like you may have found the issue, do  you still want me to generate a log file?

cprezzi, 2017-06-01:
@KirAsh4 As I was able to replicate the problem, the log is not needed anymore.

KirAsh4, 2017-06-09:
Latest release is still exhibiting this. I've attached a log file. Steps I followed to generate this log file (after adding the .env file):
a) launch LW4 (4.0.734-96-x64)
b) Switch to Comms tab and clicked Connect
c) Switch to Control tab and clicked Home All
d) Closed LW4

It's a constant stream of Offset messages as shown in my very first post.

[logfile.txt](https://github.com/LaserWeb/LaserWeb4/files/1065447/logfile.txt)

cprezzi, 2017-06-12:
@KirAsh4 What you see in the log is normal. These are the position messages that we get every 250ms from the controller, but that's not sent to the frontend. Only changes should be sent to the frontend.

KirAsh4, 2017-06-12:
Right, I only linked it because you had asked for it previously, _before you made a fix_. What I'm saying is, the problem is still there. In the front end, I'm constantly seeing these Offset messages whenever the laser is working. They seem to correlate to whenever the screen is refreshed and the blue dot indicator is being updated, but I could be wrong here.

KirAsh4, 2017-06-16:
Hey @cprezzi, can I provide you with any more information on this, or did you secretly already fix this? :)

commit c3e42cc4 (2017-05-25): Fix #324