LaserWeb/LaserWeb4 #327: Delay refreshing operation value list

cojarbi, 2017-05-29:
While changing operation type, is not until a value is entered that the corresponding list of values is refreshed. 

http://g.recordit.co/aNsqPjNJoT.gif

Latest commit build / OSX

jorgerobles, 2017-05-29:
@cojarbi Cannot reproduce error. 😮

cojarbi, 2017-05-29:
I just compiled in windows 10 and getting the same behaviour. 

Try the following:
( make sure CNC mode is on )
Load SVG
Drag file into operation 
Change between the various operation types (inside, outside, etc ) , you will not see   options change unless you type or change a value

jorgerobles, 2017-05-29:
@openhardwarecoza, @cprezzi, @tbfleming could you please test if the error arises on clean copies? I cannot reproduce. :|

openhardwarecoza, 2017-05-30:
Confirmed on Win 10 - I get this:
Open file and add document to a new operation: Defaults to 'laser cut' and parameters are as it should be for laser cut. Change operation dropdown to say, Mill Outside.   Paramater stay as they are for laser cut (for example Tool diameter etc stays missing). Start typing in any field (say, Name), suddenly renders the missing fields

openhardwarecoza, 2017-05-30:
Ruling out SVG - my test was with a DXF

openhardwarecoza, 2017-05-30:
No errors in devtools, even with debug on

jorgerobles, 2017-05-30:
I've tested on linux too.. no issues here 😞

openhardwarecoza, 2017-05-30:
I havent tested latest from dev-es6: That was with last release  (732/78)

cojarbi, 2017-05-30:
Mind that clicking on the grid or document section also causes the list to refresh.

openhardwarecoza, 2017-05-30:
Confirmed, grid click works on win10 too

openhardwarecoza, 2017-05-30:
http://recordit.co/8PER1PYvNV

jorgerobles, 2017-05-30:
I'm going to compile 733 for windows. Maybe something has been fixed.

cojarbi, 2017-05-30:
Tested on Linux Ubuntu 17.04 VM, same thing.

Could this have something to do with Display cache? On or Off does not fixe the issue but..

jorgerobles, 2017-05-30:
The only thing we could do is test from sources without compiling to binaries and go back a commit each time :sweat:

jorgerobles, 2017-05-30:
https://github.com/LaserWeb/LaserWeb4-Binaries/releases/download/v4.0.733-79/LaserWeb.Setup.4.0.733-79-x64.exe

jorgerobles, 2017-05-30:
http://recordit.co/h907TrBzs6

cprezzi, 2017-05-30:
It workes correctly on my installed 732-78-x64 on Win10. I'm installing 733 now.

cprezzi, 2017-05-30:
Version 733-79 works also as expected on Win10.

jorgerobles, 2017-05-30:
@cprezzi What gpu do you have? (if any)

cprezzi, 2017-05-30:
AMD Radeon HD 5700 Series

jorgerobles, 2017-05-30:
ok. I got nvidia.. is not that.

openhardwarecoza, 2017-05-30:
Nvidia GeForce 210 (yip from couple decades ago) and have the issue. Will
try new build when i get home

cprezzi, 2017-05-30:
Running from dev environment also works as expected.

cprezzi, 2017-05-30:
I have tested 733-78-x64 also on the Pipo X10 and it works correctly.

openhardwarecoza, 2017-05-30:
Tested: Still same, but, one little difference: When switching operations, now i can see the red warning flags appear right about where they would be, but the actual fields still just render after typing or clicking grid

cojarbi, 2017-05-30:
I do have same behavior with the above hosted live version and the above build.

jorgerobles, 2017-05-30:
Could be a Chrome issue? Versión 58.0.3029.110 (64-bit) on Mac

cojarbi, 2017-05-30:
Version 58.0.3029.110 (64-bit) Windows 10

tbfleming, 2017-05-30:
I Just got my desktop building LW again after a fresh OS install. My EVGA GTX 980 Ti crashed so many times it hosed my OS image beyond what Windows recovery could fix. Fortunately it didn't damage hardware. Yet another manufacturer is on my sh** list.

Works as expected:
* Windows 10 Version 1703
* Node 6.10.3
* Chrome 58.0.3029.110 (64-bit)
* Asus GTX 770

cprezzi, 2017-05-30:
Same here: works on Chrome 58.0.3029.110 (64-bit)

jorgerobles, 2017-05-30:
@tbfleming Ouch! :band:

cprezzi, 2017-05-30:
What about Javascript version. I remember I got a message of a Javascript update when browsing some days ago.

tbfleming, 2017-05-30:
That's tied to the browser version.

jorgerobles, 2017-05-30:
Posted branch without menuitem https://github.com/LaserWeb/LaserWeb4/tree/no_context_menu
please checkout and test your faulty ones :smile:

cprezzi, 2017-05-30:
I think I was using Firfox. Probably they do update the js engine saparately.

cojarbi, 2017-05-30:
Not working on:

- Windows 10 Pro 10.0.14393 Build 14393
- Version 58.0.3029.110 (64-bit)
- Node 6.10.0
- Intel Iris 540 16MB

cojarbi, 2017-05-30:
If ive done the process correctly, im still having the same issue with that branch :(

jorgerobles, 2017-05-30:
Then go reverting 1 week then 2 :(

tbfleming, 2017-05-30:
@jorgerobles instead of posting branches, maybe suggest specific hash revisions for @cojarbi to try.

cojarbi, 2017-05-30:
I will be reverting commits by a week range to pin down where this started

jorgerobles, 2017-05-30:
Yes, that branch was only a longshot. Will delete

cojarbi, 2017-05-30:
I just noticed that the problem is seen while changing between same machine type operations from a normal cut to a outside/inside.

If I go from any laser operation to any mill operation it works as expected. 

If I go from Laser cut to laser cut inside, it does not refresh. The same with milking ops.

I continue to looking for the origin

tbfleming, 2017-05-30:
> If I go from Laser cut to laser cut inside, it does not refresh. The same with milking ops.

Reproduced this for laser. I don't have any CNC cows to test the milking ops.

jorgerobles, 2017-05-30:
LOL. so, you have reproduced the error?

tbfleming, 2017-05-30:
yes. debugging it now.

jorgerobles, 2017-05-30:
@tbfleming Found!.. and fixed! (not commited)

jorgerobles, 2017-05-30:
operation:534, componentDidUpdate is not the place to put the code.. should be inside componentWillReceiveProps

commit eb03fafe (2017-05-30): Fix #327
closed 2017-05-30 by jorgerobles