LaserWeb/LaserWeb4 #331: lw.svg-parser: filter out useless warnings

tbfleming, 2017-06-01:
It's near impossible to import any SVG without getting "The file has minor issues." It's confusing users. We should edit lw.svg-parser to not log warnings about things we know we can safely ignore, e.g. things in the sodipodi namespace.

jorgerobles, 2017-06-01:
https://plus.google.com/105762576990882303746/posts/SXZV18S5drD?fscid=z133c3xijkrri3nos04cfvhw5sregt1rib0.1496334985536507

jorgerobles, 2017-06-01:
Yes. That is annoying. And the worst is that user cannot do anything the most of times (because are parser limits). I can put a collapsible detail of the error (as G+ user suggested), but will be useless. 
The errors were piped initially to chrome console, then to lw console, then refactored the warning to something more abstract. 

I think just to pipe all the warnings to chrome console again. If anyone complains then may open an issue and post the file. @tbfleming what do you think?

tbfleming, 2017-06-01:
We could do this:
* Keep the UI as is
* Fix lw.svg-parser to not warn about stuff that's safe to always ignore (e.g. the sodipodi stuff)

closed 2017-06-01 by jorgerobles

tbfleming, 2017-06-01:
Don't want to skip warning about all unknown tags and attrs, just the ones we know we don't need.

jorgerobles, 2017-06-01:
Ok I will revert, but then, what approach should do? Whitelisting svg
namespace? Maybe should be done before parsing.

tbfleming, 2017-06-01:
Don't warn about anything involving the sodipodi namespace or AI's namespace (don't know the name).

commit d77542a9 (2017-06-02): Fix #331 - skipped vendor warnings.