LaserWeb/LaserWeb4 #332: lw.svg-parser mishandles group transforms

tbfleming, 2017-06-03:
[group transform.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/1049850/group.transform.svg.txt)

Inkscape:
[image]

https://lautr3k.github.io/lw.svg-parser/dist/example/  :

[image]

LW4:

[image]

tbfleming, 2017-06-03:
I bypassed lw.svg-parser's transform handling. src/reducers/document.js now does it instead.

rotate branch only. dev-es6 is still broken.

jorgerobles, 2017-06-08:
As merged, fixed.

closed 2017-06-08 by jorgerobles