LaserWeb/LaserWeb4 #334: Request: Support LinuxCNC/Mach3 Machines - Details for change within

McAdamous, 2017-06-07:
This is amazing, the kind of software I have been looking for. But unfortunately it does not support some G Codes that would open the doors to a whole other load of users.

In LinuxCNC and Mach3, lasers are often used with PWM which is driven by G67 and G68 commands.

I tested out LaserWeb and I think there could be an easy way to add this option.

------------------
These are Idea Instructions:

SETTINGS:

Under machine settings make an option "Control Laser Power using G67/G68 Commands" A tooltip can suggest this is for other machines that do not use the G1 SX Commands.

If this setting is activated then the following options appear:

Enter PWM Signal Number [Number Field] (PWMSigNo)
Tooltip: If you only have one PWM device on your machine then set this number to 0, If you have more than one PWM device (for example a PWM controlled Spindle) then check your configuration.

Enter PWM Minimum Power [Number Field] (PWMMinPow)
Tooltip: If your laser requires a small amount of power to keep the diode on then please enter this amount here, otherwise set as 0

GCODE:

If the value of the Machine Setting is ON then:

Insert Laser on BEFORE move:
G67 E(PWMSigNo) Q(LaserPower)
G1 Move
G68 E(PWMSigNo) Q(PWMMinPow)

##(LaserPower) is the same value currently used for G1 SX
------------------------

G67 tells the machine to only carry out that setting once the laser starts moving
G68 tells the machine to immediately execute the command

I tried to find where the GCode is being compiled to work on this myself but my programming skills are very limited.

I can help, test and give feedback - but I know this would be a really cool addition to this program.

tbfleming, 2017-06-07:
Do many people use LinuxCNC or Mach3 with lasers? Have you tested gcode like that with a large number of small moves (e.g. rastering)? Does it run well? Stutter?

mayhem2408, 2017-06-07:
I have a stepcraft CNC machine which recently released a laser addon. The software (called UCCNC) uses a similar format to LinuxCNC for PWN control with M10/M11 used for Laser on/off

M10 Q255 ; Set laser to full power
G1 X10 Y0 ; Move and Burn
M10 Q127 ; Set Later to 50% power
G1 X20 ; Move and Burn
G0 X30 ; Rapid move without laser
M11 ; Turn laser off
G1 X0 Y0 ; Move without laser

It can feed at F3000 at 0.1mm resolution smooth as silk. It reads ahead and plans 1000 lines and like LinuxCNC/Mach3 does not suffer for USB/Serial bandwidth issues.
As with LinuxCNC/Mach3/UCCNC, it is not a simple case of changing the S to a Q. It must have its own line. I have a basic inkscape plugin that generates my GCode and it would be nice to see this in LaserGRBL.

PicEngraverToo, 2017-06-07:
This DAC only requires G1 X,Y, & Z or A, B, C axis linear commands and outputs 8bit analog modulation or PWM. It's controlled by the axis's step & direction signals. No special gcode commands needed. 
https://jtechphotonics.com/?product=dacpwm-converter-board

tbfleming, 2017-06-07:
I see 4 competing methods so far:

1. S controls the power. This is what we currently support.
2. G67 E(PWMSigNo) Q(LaserPower)
3. M10 Q(LaserPower)
4. An axis controls the power

McAdamous, 2017-06-07:
@tbfleming 
Perhaps 2+3 could be combined into one solution.
User enters command prefix (Either M67 E0 OR M10) script add in the last QPWM
The option would be:
Put Laser control in Move Commands
Put Laser Control Wrapping Move Commands

@PicEngraverToo I Think that DAC Solution is complicated... No idea why that board costs $100...

If the user really didnt have any ouput at all for PWM signal there are other ways to control a laster (analog etc) all of which should be able to be configured in whichever controller/board... But then again I dont know how hard it would be to add something custom into Mach3

PicEngraverToo, 2017-06-07:
It costs $100 because it's easy to setup with Mach3 and gives good results raster laser engraving photos. We consulted J-Tech Photonics on the design of that board. The Engraving Example on the PicConvert Page was done by me with one of our Mach3 CNC laser engravers with the DAC modulating the laser driver.    

Using an US digital MA3 10bit magnetic shaft encoder driven by a stepper or servo motor to modulate laser driver has been around the longest. The DAC using a digital potentiometer was developed by us later, but it's only 8bit. Both of those methods were prior to PWM using Z commands with grbl 8c or S commands with grbl 9g for PWM. 

I prefer the 10bit encoder myself. I have 2 Mach3 CNC laser engravers setup with encoders (one servo & 1 stepper) at my shop and they give better shading results then any other method I have tested. It's a shame grbl does not have 10bit PWM resolution.

tbfleming, 2017-06-07:
Smoothieware and grbl-lpc use the 1769's PWM, which runs on a 120 MHz clock. If you run the PWM at a 4 kHz period (user config) this gives 30,000 steps, slightly less than 15-bit resolution. Running it with a 100 kHz period gives 1200 steps, slightly more than 10-bit resolution.

cprezzi, 2017-06-07:
@mayhem2408 @PicEngraverToo Please accept that this is not a place to advertize your solutions, it's the place for LaserWeb issues only!

mayhem2408, 2017-06-08:
@cprezzi I was not advertising anything. Merely suggesting that GRBL and Smoothie are not the only motion controllers out there for laser engraving out there and that support of other solutions would be nice.

tbfleming, 2017-06-08:
We currently have these gcode settings:
* start (start of file; grbl normally has M4 S0)
* end (end of file; grbl normally has M5)
* tool on (start of a set of g1 moves; normally empty since Smoothie and grbl use g0/g1 to turn laser off and on)
* tool off (end of set of g1 moves; normally empty)
* pwm max value (default 1.0; 100.0 and 1000.0 are common values)

We could add:
* pwm min value (default 0.0)
* laser power gcode (default ```S```)
* laser power on separate line (checkbox)

@McAdamous 's case:
* tool off: ```M68 E0 Q<something>```
* pwm min value: ```<something>```
* pwm max value: ```<something>```
* laser power gcode: ```M67 E0 Q```
* checkbox on

@mayhem2408 's case:
* end: ```M11```
* pwm min value: ```<something>```
* pwm max value: ```<something>```
* laser power gcode: ```M10 Q```
* checkbox on

@PicEngraverToo 's case:
* tool off: ```<something>```; eg. ```G0 A0```
* pwm min value: ```<something>```
* pwm max value: ```<something>```
* laser power gcode: ```<something>``` eg. ```G1 A```
* checkbox on

The checkbox must be on for this final case because LW assumes that laser power changes are instantanious. e.g.

```
G1 X100 S1.0
```

Smothieware and grbl 1.1 run run the laser at power 1.0 throughout the move; they don't ramp it like an axis would.

cprezzi, 2017-06-08:
@tbfleming This could also help to support MarlinKimbra, which needs M3 S on a separate line.

arthurwolf, 2017-06-08:
> We currently have these gcode settings:
>
>    - start (start of file; grbl normally has M4 S0)
>    - end (end of file; grbl normally has M5)
>    - tool on (start of a set of g1 moves; normally empty [...]

Could you clarify exactly what you mean ? either I'm not understandinq your
meaning or this is incorrect.
Smoothie does ramp laser power proportional to speed when axes accelerate
and decelerate.

McAdamous, 2017-06-08:
@tbfleming 

I think that would work.
not sure if others require Laser Off on the line following move command or not.

@arthurwolf 

I think that is in relation to issueing Laser On commands before Movement. Laser would be at full power before the axis gets up to speed (acceleration settings)

arthurwolf, 2017-06-08:
> @tbfleming <https://github.com/tbfleming>
>
> I think that would work.
> not sure if others require Laser Off on the line following move command or
> not.
>
> @arthurwolf <https://github.com/arthurwolf>
>
> I [...]

Smoothie does have a command to manually turn the laser on.
If this is what we are talking about, if you issued that command, then a
move, what would happen is the move would still ramp following
acceleration, the G1 would essentially nullify the fire command instantly
when it starts moving.

tbfleming, 2017-06-08:
@arthurwolf: Different ramping. Make the following assumptions to see what I'm talking about:

* In the middle of a raster line
* Gcode is holding same feed throughout raster line
* Running from an SD card
* Raster line is broken up into a bunch of G1's with different S values

```
...
X150 S1.0
X151 S0.5
X152 S1.0
...
```

My understanding is that Smoothie:
* Holds power at 1.0 during first move
* Holds power at 0.5 during second move
* Holds power at 1.0 during third move

If this was Mach using A instead of S:
* Linearly ramp power from previous value to 1.0 during first move. It doesn't reach 1.0 until the end of the move.
* Linearly ramp power from 1.0 to 0.5 during second move. It doesn't reach 0.5 until the end of the move.
* Linearly ramp power from 0.5 to 1.0 during third move. It doesn't reach 1.0 until the end of the move.

We have to move the A values to a different line on Mach to prevent this behavior.

arthurwolf, 2017-06-08:
> @arthurwolf <https://github.com/arthurwolf>: Different ramping. Make the
> following assumptions to see what I'm talking about:
>
>    - In the middle of a raster line
>    - Gcode is holding same feed [...]
Smoothie does, thanks for clarifying.

> If this was Mach using A instead of S:
>
>    - Linearly ramp power from previous value to 1.0 during first move. It
>    doesn't reach 1.0 until the end of the move.
>    - Linearly ramp power from 1.0 to [...]

tbfleming, 2017-06-10:
New settings in linuxcnc branch. Please test.
* Laser Intensity: defaults to S
* Intensity Separate Line: checkbox
* PWM Min S Value

The Laser Cut and Laser Fill Paths operations support these. Laser Raster doesn't yet. The Laser Preview (black line) doesn't show up with gcode that doesn't use S for intensity.

McAdamous, 2017-06-12:
Hmmm... Spent quite a while to see how to install a different branch under windows and didnt find anything...
Any pointers?

tbfleming, 2017-06-12:
Assuming Windows, with git and node already installed and in your path:

```
npm install --global windows-build-tools
git clone https://github.com/LaserWeb/LaserWeb4.git
cd LaserWeb4
git checkout linuxcnc
npm run installdev
npm start
```

McAdamous, 2017-06-12:
Ok, I updated to the Linuxcnc Branch. Made some GCode and so far it looks great.

I will do some actual proper testing when I get to the Workshop tomorrow - So for LinuxCNC it works perfectly fine.

Maybe @mayhem2408 's Test Input would be valuable

McAdamous, 2017-07-07:
Since no one else has given any feedback, whats the chances of this being merged? :D

tbfleming, 2017-07-07:
How did your testing go?

mayhem2408, 2017-07-07:
@tbfleming I've always had problems trying to compile the sources of laserweb and unfortunately do not have the time to try and figure it out. Is anyone able to give me a clear step by step on how to do this?

tbfleming, 2017-07-07:
@mayhem2408 did you try the instructions above?

mayhem2408, 2017-07-08:
@tbfleming Just tried it again. The first two attempts had git return an error
```
ffatal: The remote end hung up unexpectedly
atal: early EOF
fatal: index-pack failed

```
Third attempt seemed OK. I'll let you know how it goes.

mayhem2408, 2017-07-08:
This is the error I get when I run "npm run installdev"

```
> laserweb@4.0.734 installdev C:\LaserWeb4
> git submodule init && git submodule update --remote && npm install && npm update lw.comm-server

-
> serialport@4.0.7 install C:\LaserWeb4\node_modules\lw.comm-server\node_modules\serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/d
ownload/4.0.7/serialport-v4.0.7-node-v46-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@4.6.1 (node-v46 ABI) (falling back to sourc
e compile with node-gyp)

C:\LaserWeb4\node_modules\lw.comm-server\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Program Fi
les\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" clean )  else (node "" clean
)
gyp ERR! clean error
gyp ERR! stack Error: EPERM: operation not permitted, unlink 'C:\LaserWeb4\node_modules\lw.comm-server\node_modules\seri
alport\build\Release\obj\serialport\serialport.obj'
gyp ERR! stack     at Error (native)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node
-gyp\\bin\\node-gyp.js" "clean"
gyp ERR! cwd C:\LaserWeb4\node_modules\lw.comm-server\node_modules\serialport
gyp ERR! node -v v4.6.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd clean' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\LaserWeb4\node_modules\lw.comm-server\node_modules\serialpor
t\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:829:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 10.0.15063
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\LaserWeb4\\node_modules\\lw.comm-server\\node_modul
es\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\LaserWeb4\node_modules\lw.comm-server\node_modules\serialport
node-pre-gyp ERR! node -v v4.6.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute 'node-gyp.cmd clean' (1)
npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "ins
tall"
npm ERR! node v4.6.1
npm ERR! npm  v2.15.9
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\LaserWeb4\npm-debug.log

npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "run
" "installdev"
npm ERR! node v4.6.1
npm ERR! npm  v2.15.9
npm ERR! code ELIFECYCLE
npm ERR! laserweb@4.0.734 installdev: `git submodule init && git submodule update --remote && npm install && npm update
lw.comm-server`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the laserweb@4.0.734 installdev script 'git submodule init && git submodule update --remote && npm in
stall && npm update lw.comm-server'.
npm ERR! This is most likely a problem with the laserweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     git submodule init && git submodule update --remote && npm install && npm update lw.comm-server
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs laserweb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls laserweb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\LaserWeb4\npm-debug.log
```

mayhem2408, 2017-07-08:
@tbfleming I've had to update my installation of node.js.

mayhem2408, 2017-07-08:
This is as far as I can get. four attempts now.
[image]

mayhem2408, 2017-07-08:
This is the console error
```
PS C:\laserweb4> npm start

> laserweb@4.0.734 start C:\laserweb4
> npm-run-all -p -r start-app start-server

> laserweb@4.0.734 start-app C:\laserweb4
> webpack-dev-server --progress --colors --open

> laserweb@4.0.734 start-server C:\laserweb4
> node node_modules/lw.comm-server/server.js

***************************************************************
        ---- LaserWeb Comm Server 4.0.114 ----
***************************************************************
  Use  http://192.168.56.1:8000 to connect this server.

* Updates:
  Remember to check the commit log on
  https://github.com/LaserWeb/lw.comm-server/commits/master
  regularly, to know about updates and fixes, and then when ready
  update accordingly by running git pull

* Support:
  If you need help / support, come over to
  https://plus.google.com/communities/115879488566665599508
***************************************************************
                                                                                                   70% 1/1 build modules
 http://localhost:8080/webpack-dev-server/
webpack result is served from /
content is served from C:\laserweb4\dist                                                           52% 5/7 build modules
webpack: wait until bundle finished: /                                                           48% 11/17 build modules
webpack: wait until bundle finished: /                                                          15% 23/234 build modules
webpack: wait until bundle finished: /                                                         46% 267/442 build modules
webpack: wait until bundle finished: /                                   90% optimize assetsMergetJsonsWebpackPlugin emi
t...                                                                                                Hash: 179e7035fe3261
c7c52d
Version: webpack 1.15.0
Time: 25501ms
                                 Asset     Size  Chunks             Chunk Names
 fa2772327f55d8198301fdb8bcfc8158.woff  23.4 kB          [emitted]
        01e78f7d1b979b7bbb0b.worker.js  68.9 kB          [emitted]
        10d409696850cb9dbab5.worker.js   613 kB          [emitted]
    10d409696850cb9dbab5.worker.js.map   731 kB          [emitted]
        bbaf5ec650d4492f457d.worker.js  1.44 MB          [emitted]
    bbaf5ec650d4492f457d.worker.js.map  1.86 MB          [emitted]
        142b4bcb50839505f554.worker.js  1.44 MB          [emitted]
    142b4bcb50839505f554.worker.js.map  1.86 MB          [emitted]
  f4769f9bdb7466be65088239c12046d1.eot  20.1 kB          [emitted]
448c34a56d699c29117adc64c43affeb.woff2    18 kB          [emitted]
    01e78f7d1b979b7bbb0b.worker.js.map   106 kB          [emitted]
  e18bbf611f2a2e43afc071aa2f4e1512.ttf  45.4 kB          [emitted]
  89889688147bd7575d6327160d64e760.svg   109 kB          [emitted]
  674f50d287a8c48dc19ba404d20fe713.eot   166 kB          [emitted]
af7ae505a9eed503f8b8e6982036873e.woff2  77.2 kB          [emitted]
 fee66e712a8a08eef5805a46892932ad.woff    98 kB          [emitted]
  b06871f281fee6b241d60582ae9369b9.ttf   166 kB          [emitted]
  912ec66d7572ff821749319396470bde.svg   444 kB          [emitted]
                              index.js  8.61 MB       0  [emitted]  main
                          index.js.map  10.7 MB       0  [emitted]  main
      ./src/data/machine-profiles.json  4.58 kB          [emitted]
chunk    {0} index.js, index.js.map (main) 8.31 MB [rendered]
    [0] multi main 64 bytes {0} [built]
    [1] (webpack)-dev-server/client?http://0.0.0.0:8080 4.16 kB {0} [built]
    [2] ./~/url/url.js 23.3 kB {0} [built]
    [3] ./~/url/~/punycode/punycode.js 14.6 kB {0} [built]
    [4] (webpack)/buildin/module.js 251 bytes {0} [built]
    [5] ./~/url/util.js 314 bytes {0} [built]
    [6] ./~/querystring/index.js 127 bytes {0} [built]
[... 1421 lines cut ...]
        [7] ./~/gl-matrix/src/gl-matrix/mat2d.js 12.3 kB {0} [built]
        [8] ./~/gl-matrix/src/gl-matrix/mat3.js 19.9 kB {0} [built]
        [9] ./~/gl-matrix/src/gl-matrix/mat4.js 67.1 kB {0} [built]
       [10] ./~/gl-matrix/src/gl-matrix/quat.js 16.1 kB {0} [built]
       [11] ./~/gl-matrix/src/gl-matrix/vec3.js 19.6 kB {0} [built]
       [12] ./~/gl-matrix/src/gl-matrix/vec4.js 15.1 kB {0} [built]
       [13] ./~/gl-matrix/src/gl-matrix/vec2.js 13.9 kB {0} [built]
       [14] ./src/lib/mesh.js 19.7 kB {0} [built]
       [15] ./~/poly2tri/src/sweepcontext.js 14.8 kB {0} [built]
       [16] ./~/poly2tri/src/pointerror.js 1.2 kB {0} [built]
       [17] ./~/poly2tri/src/xy.js 3.07 kB {0} [built]
       [18] ./~/poly2tri/src/point.js 6.98 kB {0} [built]
       [19] ./~/poly2tri/src/triangle.js 16.1 kB {0} [built]
       [20] ./~/poly2tri/src/sweep.js 25.6 kB {0} [built]
       [21] ./~/poly2tri/src/assert.js 766 bytes {0} [built]
       [22] ./~/poly2tri/src/advancingfront.js 4.04 kB {0} [built]
       [23] ./~/poly2tri/src/utils.js 2.55 kB {0} [built]
       [24] ./~/snapsvg-cjs/dist/snap.svg-cjs.js 251 kB {0} [built]
       [25] ./~/eve/eve.js 13.4 kB {0} [built]
       [26] ./~/script-loader!./~/web-cam-cpp/web-cam-cpp.js 190 bytes {0} [built]
       [27] ./~/script-loader/addScript.js 234 bytes {0} [built]
       [28] ./~/raw-loader!./~/web-cam-cpp/web-cam-cpp.js 617 kB {0} [built]
webpack: Failed to compile.

```

jorgerobles, 2017-07-08:
Closer! @mayhem2408 

You have to `npm run installdev` once before start.
Laserweb depends on two submodules lw.machines and lw.materials to run.

mayhem2408, 2017-07-08:
@jorgerobles I have run it many times. I get the following.
```
PS C:\laserweb4> npm run installdev

> laserweb@4.0.734 installdev C:\laserweb4
> git submodule init && git submodule update --remote && npm install && npm update lw.comm-server

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules\chokidar\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.2: wanted {"os":"darwin","arch":"any
"} (current: {"os":"win32","arch":"x64"})
laserweb@4.0.734 C:\laserweb4
`-- lw.comm-server@4.0.114  (git+https://github.com/LaserWeb/lw.comm-server.git#ac029d4ac713a4b1f301de3b90e781dd7276e3ab
)

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules\chokidar\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.1.2: wanted {"os":"darwin","arch":"any
"} (current: {"os":"win32","arch":"x64"})
PS C:\laserweb4>
```

jorgerobles, 2017-07-08:
Look at data folder. Is there not lw.machines nor lw.materials folder?

mayhem2408, 2017-07-08:
@jorgerobles 
Yes they are there.
```
PS C:\laserweb4\src\data> dir

    Directory: C:\laserweb4\src\data

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        8/07/2017   7:17 PM                lw.machines
d-----        8/07/2017   7:17 PM                lw.materials
-a----        8/07/2017   7:16 PM            302 macros.json

PS C:\laserweb4\src\data>
```

jorgerobles, 2017-07-08:
Then a webpack plugin is missing or not working.

Ensure `MergeJsonPlugin` is on `webpack.config.js`

```
 plugins: [
        new MergeJsonWebpackPlugin({
            "output": {
                "groupBy": [
                    {
                        "pattern": "./src/data/lw.machines/machines/*.json",
                        "fileName": "./src/data/machine-profiles.json"
                    }
                ]
            }
        }),
   ...
```

and  `"merge-jsons-webpack-plugin": "1.0.7",` on `package.json` `devDependencies`

mayhem2408, 2017-07-08:
My package.json had `"merge-jsons-webpack-plugin": "^1.0.6",`
I changed it to 1.0.7 and it compiled successfully.

But how when I go to http://localhost:8000/ I get an error in the console
`ERROR:  webServer error:/ :  Not Found`

jorgerobles, 2017-07-08:
hmm I was not expecting that one :smile:  📢  @cprezzi 📢 
@mayhem2408 please run `npm run start-server` on a separate console and report.

mayhem2408, 2017-07-08:
```
PS C:\WINDOWS\system32> cd \laserweb4
PS C:\laserweb4> npm run start-server

> laserweb@4.0.734 start-server C:\laserweb4
> node node_modules/lw.comm-server/server.js

events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE :::8000
    at Object.exports._errnoException (util.js:1018:11)
    at exports._exceptionWithHostPort (util.js:1041:20)
    at Server._listen2 (net.js:1258:14)
    at listen (net.js:1294:10)
    at Server.listen (net.js:1390:5)
    at Object.<anonymous> (C:\laserweb4\node_modules\lw.comm-server\server.js:141:5)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)

npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "run
" "start-server"
npm ERR! node v6.11.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! laserweb@4.0.734 start-server: `node node_modules/lw.comm-server/server.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the laserweb@4.0.734 start-server script 'node node_modules/lw.comm-server/server.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the laserweb package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node node_modules/lw.comm-server/server.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs laserweb
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls laserweb
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\laserweb4\npm-debug.log
PS C:\laserweb4>
```

mayhem2408, 2017-07-08:
```
PS C:\laserweb4> npm --version
3.10.10
PS C:\laserweb4> node --version
v6.11.0
```

jorgerobles, 2017-07-08:
Then you have the server already running. Go to win task manager and kill some node's.
Then run `npm start`

mayhem2408, 2017-07-08:
I've killed off all node's and ran npm start again. It says compiled successfully. Still not working though.

jorgerobles, 2017-07-08:
:frowning_face:  Zip your folder and share on drive. Maybe overlook something.

mayhem2408, 2017-07-08:
@jorgerobles The Zip file is over 100Mb. Where do you want it uploaded to?

jorgerobles, 2017-07-08:
Google drive, dropbox, box... whatever it fits and is somewhat persistent to let the devs check

mayhem2408, 2017-07-08:
@jorgerobles Upload speeds are horrible here. Dropbox say 1 hour to upload. I'll post the link once it's done.

mayhem2408, 2017-07-08:
Here is the zip file.
https://www.dropbox.com/s/94ylum48hyikpt9/LaserWeb4.zip?dl=0

cprezzi, 2017-07-10:
The LW4 frontend is hosted on port 8080!
Port 8000 is the websocket backend.

cprezzi, 2017-07-10:
After `npm start`, your default browser should open with url `localhost:8080/webpack-dev-server`. Please set chrome as default browser, or you will not be able to use all features.

McAdamous, 2017-08-10:
Ok, did anyone else test this?
I'm still using the manual install which never saves my settings and is kind of putting me off using this otherwise amazing app.
Anything else I can test with?

tbfleming, 2017-08-10:
@McAdamous I didn't realize you tested it in the shop; I was waiting for confirmation that someone had. I'll merge it next time I'm working on the sources.

McAdamous, 2017-08-10:
@tbfleming Yup, been using it for a long time now, works perfectly fine for me with LinuxCNC

tbfleming, 2017-08-10:
It's now in the mainline. I also deployed it to the live version: https://laserweb.github.io/LaserWeb4/dist/

closed 2023-08-29 by McPizza0