LaserWeb/LaserWeb4 #335: Add rotation arrows

cojarbi, 2017-06-08:
I know you guys just finished this but maybe the following or similar could be added either 45 or 90 degrees interval?
[image]

tbfleming, 2017-06-08:
I'm thinking it might be cleaner than what I currently have to do this: Changing the value won't immediately rotate. Instead it will change the increment that the arrows rotate the doc. It could default to 45.

jorgerobles, 2017-06-08:
[image]

closed 2017-06-16 by cojarbi