LaserWeb/LaserWeb4 #336: QCad DXF, Saved R12, all lines exploded, edges not connecting

PeterGrace, 2017-06-08:
[TeacherStandsFullSheet.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/1061799/TeacherStandsFullSheet.dxf.txt)
[LW4-TeacherStandsFullSheet.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1061802/LW4-TeacherStandsFullSheet.json.txt)

Hello everyone (and especially @iceblu3710)

I've attached a dxf, and a workspace, where I'm trying to cut acrylic sheets to make teacher gifts.  In QCad, I exploded all of the shapes, and they show connected in QCad, however when viewed in the LW4 workspace, the lines do not connect.

I vaguely remember a few weeks ago we were discussing a similar issue, and the issue was remedied by saving as R12, however this time around it appears that the issue persists.  If I instead connect the two edges via a polyline, the vertices connect, but the curve is replaced by a set of straight lines.

Is this a problem with QCad's generation of the R12 DXF, or with the LW4 parser?  Any suggestions on how I can ensure the lines are connected, if this is something that can't be fixed with a LW4 change?

Thanks!
-pete

iceblu3710, 2017-06-08:
I'll look into it and let you know.

iceblu3710, 2017-06-09:
> In QCad, I exploded all of the shapes, and they show connected in QCad, however when viewed in the LW4 workspace, the lines do not connect. 

If you explode a square you get 4 lines. The vertices may line up but the program treats them as four separate entities.

> If I instead connect the two edges via a polyline, the vertices connect, but the curve is replaced by a set of straight lines.

That is normal, If the entity is not a polyline bulge then it will likely be exported as a series of polylines that approximate the curve.

> Is this a problem with QCad's generation of the R12 DXF, or with the LW4 parser? 

Can I see the original Qcad file? When I open the provided DXF the objects are indeed exploded and entirely seperate, only way to connect is cad commands like EDITPOLYLINE.

PeterGrace, 2017-06-09:
[TeacherStandsFullSheet-polylines.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/1062806/TeacherStandsFullSheet-polylines.dxf.txt)

Here's the file with them converted back to polylines.

iceblu3710, 2017-06-09:
> And I should be more clear, sorry -- the rectangles I wasn't concerned about, it was the edges on the curves that weren't connecting with the straight lines that had me scared.

Ahh, I see it now. It doesn't seem to be consistent and only related to the ARC entities. Will run some tests.

iceblu3710, 2017-06-09:
Ok, some weird results and findings.  

If you remove the frame from around the whole drawing all the curves work fine in the polyline file (saved as r12 all the way to 2013 using draftsight). That only solves the curves rendering properly.  

The missing bits on the drawings are due to the fact the parser is classifying them as `shapes: false` which stems from `70: 0(default), 1(shape), 128(peditline)`. ([Spec, Page 102](http://images.autodesk.com/adsk/files/autocad_2012_pdf_dxf-reference_enu.pdf)) If you open the DXF in a text editor and search for `polyline` you will see the 70 descriptor. What is odd is >128 is not a valid setting, the file ends the peditline shape by +1 to 128 to get 129 which is illegal and is defaulting to 0 which is `shape: false`

I use the `shape: true` value to push the last vertex back to the first value to "close off" the shape. I will report the issue to the dxfparser github page as I have seen a few issues with 129 popping up. Possibly forcing an if >128 then -128 could solve a lot of these issues. Will post again when I get more time.

PeterGrace, 2017-06-10:
> If you remove the frame from around the whole drawing all the curves work fine in the polyline file (saved as r12 all the way to 2013 using draftsight). That only solves the curves rendering properly.

Removing the frame, saving as R12 in QCad results in straight lines, not curves, for me.  What version are you using?
[image]

iceblu3710, 2017-06-11:
Well, the unconnected segments issue has been fixed. Waiting on a version bump on the dxf-parser library to take effect.  

I think qCad may be exporting the DXF with some strange entity layering or possibly be relying on "layer styles" which is wholly unsupported. When I load your original polyline file into DraftSight and save it as R12 it loads without the curve information (DraftSight makes no changes as there are no R12 incompatible features) but when you delete or modify the drawing (such as remove the frame) then it completely re-writes the file to R12 "dumbed down" standards.

I will install qCad and see if there is a setting somewhere causing this, will likely need to document an official supported settings page on the [cncpro.co](http://cncpro.co/index.php/35-documentation/working-with-files/working-with-dxf) site.

---

**NOT RECCOMENDED** 
_If you want to manually update and can't wait then you can remove the /nome_modules/dxf-parser and `git clone https://github.com/gdsestimating/dxf-parser.git` the current master branch. This may muck up future updates so once an update is posted you can remove the folder again and do the `npm run-script installdev`_

---

[image]

iceblu3710, 2017-06-22:
PR #352 bumps the dxf-parser a minor version and brings in full bitfield support for entity options. Every instance should properly render as a shape as long as the program doing the exporting exports shapes.

I have not been home in a week so have not been able to install QCad. I read it uses DXFLib as its backend so it's susceptible to outputting poorly structured DXF as the write calls dump to a file without internal structure checking.

iceblu3710, 2017-06-23:
I took a look through QCad and nothing jumped out at me for settings. However whenever I save your polyline file as `R15 (2000/LT2000) DXF Drawing(dxflib) (*.dxf)` it loads in LaserWeb properly now.

I will keep looking into QCad as I have time but it looks like it handles layer "0" differently than other programs. 
[image]

PeterGrace, 2017-06-23:
Thanks for your help on this issue!  From your investigation, should I be saving layer 0 in "Best compatibility" mode or behaves-like-a-regular-layer?

iceblu3710, 2017-06-23:
> Thanks for your help on this issue! From your investigation, should I be saving layer 0 in "Best compatibility" mode or behaves-like-a-regular-layer?

Neither work the way I would expect...

PR #353 fixes the issue of the entities not touching. I had to comb through the math and code to find my mistake. This will let you use exploded polylines via QCad (your file in post 1 works correctly now). Polylines work, just need to find what QCad setting is making them weird.

Do you have the pro version of QCad? I wonder if you put the drawing on a new layer if it will work.

closed 2017-06-23 by tbfleming

iceblu3710, 2017-06-23:
@PeterGrace 

I have pushed a PR to the dxf-parser author to fix the QCad polylines not curved issue. dxflib uses a different nomenclature of POLYLINE & VERTEX instead of a simple LWPOLYLINE and the bulge geometry was being ignored on plane VERTEX's. That should be all we need to make QCad work out of the box!

iceblu3710, 2017-07-10:
@PeterGrace 

Another version bump on the parser has solved all of the import issues. Both your files in this thread load properly now. (#375 )

PR #353 (2017-06-23): Fix arc vertices math
Closes #336
commit 2e23efe9 (2017-06-23): Fix arc vertices math

Closes #336