LaserWeb/LaserWeb4 #338: create a multi job array

jtronics, 2017-06-08:
I'm trying to laser mutiple copies of the same job at once. 
How can I create a job array like in the picture. In moment I do it manual, by copying the different files together.
When lasering multiple objects there should be two modes like in cura for 3d printing): 
- "laser-at-once"
- "one-at-a-time"
Thanks Martin
[image]

cprezzi, 2017-06-09:
I have a solution for this (kind of).

First, create the folowing macros:
- shiftZeroRight
  - G92X-80 (the distance to shift in x)
  - G0X0Y0
- shiftZeroLeft
  - G92X80
  - G0X0Y0
- shiftZeroUp
  - G92Y-50 (the distance to shift in y)
  - G0X0Y0
- shiftZeroDown
  - G92Y50
  - G0X0Y0

These macros move the origin right/left/up/down by the defined distances and then moves the head to 0/0. This way you can just move the origin and run the job again.

tbfleming, 2017-06-09:
I wonder if we should add gcode's looping constructs to the sender.

jorgerobles, 2017-06-09:
I was wondering about svg symbol and redux proxies...

cprezzi, 2017-06-09:
@tbfleming That could be interesting.

pedjas, 2020-08-16:
Three year later and seems there is no better solution. I am seconding this request.

It really would be helpful to set cutting rules for an object and then just set pattern to cut number of copies.

cprezzi, 2020-08-17:
@pedjas You are welcome to implement that and make a pull request.

pedjas, 2020-08-20:
> 
> 
> @pedjas You are welcome to implement that and make a pull request.

So I have to be programmer not a user to be able to use LaswerWeb?

cprezzi, 2020-08-24:
If this feature would be widely demanded (and a devs had time to develop it), it would have been made. But as you see, you are just the second one who asked for it in more than 3 years. That's not realy a big demanded.

LaserWeb is not actively developed anymore, because most of the devs left the project. I just try to maintain it as good as I can and I am open to let other devs work on it too.

Talking about attitude: It's also not a very pleasant attitude to start your fist post in this issue with a criticism: "Three year later and seems there is no better solution."

cprezzi, 2020-12-07:
@jamescbury I did something like this with two makros. First I made sure the job gcode had a move to 0,0 at the end (with "End Gcode" setting). Then I made a macro1 that moved x mm in x direction and reset the origin. Macro2 moved the origin back to left border and x mm um in y. This way I could just hit macro1 or macro2 between each job run.