LaserWeb/LaserWeb4 #339: Operation - Laser Fill Path - Text/Word

jtronics, 2017-06-08:
If I fill a vector text (svg file), so it is done one symbol after another. Is it possible to add the option to do it like in the raster-function.  Lasering the whole word at once.

In moment I use Inkscape and my way is ... I export the text as png and than reimport the png to Inkscape, so that it is a embedded raster image. So i have only one source file for the whole job, including the raster and vector things.

Is it also possible to use the svg text object? 
In moment I have to convert the Text in to path objects.

cojarbi, 2017-06-08:
@jtronics 
Adding the option to do the whole would at once would be the same as Raster, why not use that?

Text should be converted as objtects for them to be readable in LW as vectors, unless you import them as images but you will only be able to Raster them.

cojarbi, 2017-06-08:
Please this kind of support questions are to be posted in our G+ community

https://plus.google.com/communities/115879488566665599508

closed 2017-06-08 by cojarbi