LaserWeb/LaserWeb4 #340: Delete Key deletes document during position editiong, instead of digit

cprezzi, 2017-06-09:
@jorgerobles I found a bug: When editing the position/size of an object, pressing the delete key deletes the whole document intead of the digit behind the cursor. The document tree is not updatet (document still visible but not usable).

jorgerobles, 2017-06-09:
F**k

jorgerobles, 2017-06-09:
Quick fix.. ctrl+delete?

closed 2017-06-09 by jorgerobles

cprezzi, 2017-06-09:
Easy, but needs to be told to people that already use del to delete a document. ;)

jorgerobles, 2017-06-09:
Yep it's on release page.  Will document ASAP

cojarbi, 2017-06-30:
Delete key on the mac not working at all to delete a path from the grid. Neither any other key combinations with delete key works

jorgerobles, 2017-06-30:
@cojarbi mouse pointer should be over workspace. (not over floating controls)- I've tested and works ok.

cojarbi, 2017-06-30:
@jorgerobles got it to work but its not straightforward on the mac since the alt key is available by pressing the fn key , so its fn+alt+delete

jorgerobles, 2017-06-30:
Key 'option' is not working?

cojarbi, 2017-06-30:
nope

jorgerobles, 2017-06-30:
O_O
https://developer.mozilla.org/es/docs/Web/API/KeyboardEvent/keyCode
option should be behave as alt :|

jorgerobles, 2017-06-30:
@cojarbi  use Command+Backspace :D

cojarbi, 2017-06-30:
@jorgerobles on Mac laptops there is no other key besides delete, which is the same as backspace

commit 9e16fe2c (2017-06-09): Fix #340 Changed “del” to “alt+del”