LaserWeb/LaserWeb4 #342: Non Squareness Compensation

OzRon, 2017-06-11:
No machine is perfect.  To be able to measure any mechanical out of square on the gantry and then compensate for it in the software is a high value feature to improve accuracy.  This feature has been implemented in OpenPnP so it should be relatively easy to incorporate into LaserWeb.  Keep up the great work!  Cheers Ron

tbfleming, 2017-06-11:
Pick and place needs a higher accuracy than what most LW users need.

OzRon, 2017-06-12:
I am glad you agree some users would benefit from this feature.  We are cutting aircraft parts and require the maximum possible accuracy achievable.  Near enough is not good enough for us as it may be for some.  No need to re-invent the wheel here, it has already been done and works well in the open source example given.

domenic-d, 2017-06-14:
We achieve this by tweaking the number of steps per rotation for the x & y motors.
In smoothieware, we adjust the config file accordingly.

OzRon, 2017-06-14:
Hi Domenic,
Thanks for your help here.  So if you measure a mechanical run out of a Laser gantry of say 2 degrees, (88/92) how would you adjust for this in Smoothieware?  Is this different to how they managed it OpenPnp?  Cheers Ron

domenic-d, 2017-06-15:
@OzRon We usually make these adjustments with a little trial and error, but i'm sure there is a mathematical way you can determine this more precisely.

We engrave a large rectangle and measure it for discrepancies.

With smoothieware you can alter the following parameters in the config file, located on the onboard sd card. Changes will be implemented after a reboot. 

```
alpha_steps_per_mm                           200              # Steps per mm for alpha stepper
beta_steps_per_mm                            200              # Steps per mm for beta stepper
```

I don't know how it is done in OpenPnp, but I suspect it is more integrated into the code generation rather than tweaking the hardware as we are doing.

mayhem2408, 2017-06-15:
@OzRon Are you referring to a square with 4x 10cm sides is not exactly 10cm, or are you referring to the corners not being 90degrees each? Altering the steps will not fix the latter.

arthurwolf, 2017-06-15:
There is an undocumented solution in Smoothie to support X/Y axes with a
non-90 degree angle :
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/robot/arm_solutions/RotatableCartesianSolution.cpp
However, nobody has ever used it ( outside of testing ), and if you think
you need it, you are doing things incredibly wrong.
Just fix your axis !! It should have a 90 degrees angle, and using software
to adjust for it is an incredibly wrong solution. It's not how you are
supposed to do things ( fix the hardware ! ) and it will lead to a lot of
problems with artefacts, moires etc down the line.
Just fix the hardware.
Having your angles square is not that hard, just fix the hardware :)

OzRon, 2017-06-15:
Hi Mayhem,
The not 90 degree corners due mechanical error in gantry alignment.  Sometimes due to the way the machine is made it is extremely difficult to correct in hardware.  Applying a correction much the same as the way a 3D printer can compensate for bed leveling error in software is how they do it in OpenPnP.  
Here is the link to the driver:  [https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/machine/reference/driver/GcodeDriver.java](url)
Definition at line 154, Algo lines 501 - 514, 1032

Thanks Domenic, I don't believe Smoothie can compensate for non-squareness but maybe it could and then benefit many users.  Will check with Arthur and advise.
Cheers Ron

tbfleming, 2017-06-15:
I have yet to see a hardware design that can't be shimmed to correct squareness, unless it has much worse problems (non-straight slides or wobbly frame).

OzRon, 2017-06-15:
Isn't it funny how when a simple software solution (proven) offered as a feature to include in a project receives more energy to defeat than actually implement?  Amazing.  

Here we have computer power at our command, but are supposed to revert to stone age solutions?

I like this to asking my teenage son to walk the dog.  He would rather argue he has too much homework (AKA gaming) to do for more time & effort than actually walking the dog.  Go figure.

Hey Arthur, 
Machining new gantry ends means two days plus solid work to fix this.  

This simply means time I cannot spend on the Smoothie II expansion boards as I was planning to use the Laser for finishing the PCB's.  I was trying to improve the quality of the prototypes.  Thanks for your help here.

Cheers Ron

arthurwolf, 2017-06-15:
Three thing :
1) It's much less work discussing this than implementing it, and discussing
something before deciding whether implementing it is *normal*
2) This is actually implemented in Smoothie, so your complaint is moot
3) The rejection of the idea has nothing to do with how much work it is,
but has to do with two things : first it's not the proper way of doing
things, and second because of this, doing it this way will have lots of
negative and unintended consequences such as laser engraving artefacts.

OzRon, 2017-06-15:
Very happy to discuss and thanks for your swift response. :-)
Please expand the Smoothie non squareness functionality and I am not complaining, but was simply trying to offer a feature enhancement.

Then there is an ideal world we would like to live in and I really like that.  

Suddenly reality sets in and doing your best to get into the ideal world it nearly gets there but not quite.
As for the collateral, like all new features, default opt out.  What am I missing?
Cheers Ron

tbfleming, 2017-06-15:
> Isn't it funny how when a simple software solution (proven) offered as a feature to include in a project receives more energy to defeat than actually implement? Amazing.

> What am I missing?

Pointing to an app written in a foreign language (Java) which browsers don't support doesn't tell us anything. Even if it was written in JavaScript, its architecture would be completely unrelated to ours and the code wouldn't port. We really would have to "re-invent the wheel here."

arthurwolf, 2017-06-15:
I linked to it before a few messages ago :
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/robot/arm_solutions/RotatableCartesianSolution.cpp
Smoothie supports rotating arm solutions, but we don't plan on documenting
or officially supporting this feature.
There are lots of things for which software solutions could exist ( like
backlash for example ), but users don't realize that using software instead
of hardware to solve those issues has a lot of problems users don't
expect/think about.
These problems are particularly worse for something like laser cutting
where any small artefact is instantly extremely visible.
Really really not a good idea, there is a reason why this isn't a common
feature anywhere, and if you've even worked with a system that has this
feature, documentation always says : don't use it, it's bad.

tbfleming, 2017-06-15:
> and if you've even worked with a system that has this feature, documentation always says : don't use it, it's bad

This reminds me of the bad-old days when people tried to do backlash compensation in SW. It wouldn't surprise me if new machine owners keep repeating that mistake.

OzRon, 2017-06-15:
Thanks Arthur.  I greatly appreciate this perspective.  Perhaps Jason NV should be enlightened?  Could you please pass this perspective on to OpenPnP as I am sure they will be very interested in why something that works should not. 

Now, Very sorry if Java is foreign?  Thought it was universal non platform specific.  Very sorry if you are now being left behind, but not difficult to evolve.

If you can measure mechanical error in the real world and correct for it in the computer world, we are the masters.

Otherwise, we become the servants to computers, not the masters.

Think about it.

Ron

arthurwolf, 2017-06-15:
> Thanks Arthur. I greatly appreciate this perspective. Perhaps Jason NV
> should be enlightened? Could you please pass this perspective on to OpenPnP
> as I am sure they will be very interested in why something that works
> should not.
See, you are missing something I've been saying : this type of compensation
has problems, but they are *much much* less a problem on a pnp than they
are on a laser. a pnp just places components at a particular place. A laser
traces or rasters, and that can "show" problems much more strongly.

> Now, Very sorry if Java is foreign? Thought it was universal non platform
> specific. Very sorry if you are now being left behind, but not difficult to
> evolve.
>
> If you can measure mechanical error in [...]

tbfleming, 2017-06-15:
>  Thought it was universal non platform specific. Very sorry if you are now being left behind, but not difficult to evolve.

That view is so 1990's. All 4 major browsers dropped Java support. It's dead, except on server.

OzRon, 2017-06-15:
Arthur, I am amazed at the resistance here to a real world problem solution already implemented by example.

Now I am shocked that Java is apparently Obsolete.

Has anyone told Jason?

:-(

arthurwolf, 2017-06-15:
> Arthur, I am amazed at the resistance here to a real world problem
> solution already implemented by example.

You are not listening, there are problems with this solution that will be
extremely problematic on a laser but that will not be visible on a pnp ...
The resistance is not to the idea, it is to the problems the idea will
cause.

> Now I am shocked that Java is apparently Obsolete.
>
> Has anyone told Jason?
>
> :-(

OzRon, 2017-06-15:
It is a simple math calculation.
Command move to X,Y point from X,Y starting point. 

Plot solution. 

Calculate theoretical world G-Codes.

Apply machine squareness error bias............ (topic of discussion)

Amazingly, within the normal travel range of the machine it lands at the exact 90 degree cut start point.

What an amazing world we live in where anything is possible!

Cheers, Ron

arthurwolf, 2017-06-15:
If you don't listen I'm going to give up trying to explain ... I'm not
saying it's difficult to do, I'm saying if you do it, some of the things
users use laser cutters for will be broken. There will be defects in
engraven lines or bitmaps etc. They will be ugly. You won't want to use
your machine because it won't do a good job.

OzRon, 2017-06-15:
OK, All so ears! 
I am very happy to learn from you.

By all means please expand why smoothie can't calculate a simple axis offset factor. Specifically.

Ron.

tbfleming, 2017-06-15:
> OK, All so ears!
> I am very happy to learn from you.
> 
> By all means please expand why smoothie can't calculate a simple axis offset factor. Specifically.

No, you're not. If you were all ears you'd know that he said Smoothie *can* do it, but that it creates real-world problems.

OzRon, 2017-06-15:
Such as all machines have to have an exact 90 degree angle mechanically built X / Y as software cannot possibly compensate?  Have I got it now?  Thanks Ron

tbfleming, 2017-06-15:
Oops. I just checked the math on https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/robot/arm_solutions/RotatableCartesianSolution.cpp and it only rotates. It doesn't correct for non-90-degree angles.

> Such as all machines have to have an exact 90 degree angle mechanically built X / Y as software cannot possibly compensate? Have I got it now? Thanks Ron

Pretty much. Software compensating for that can create real-world issues.

arthurwolf, 2017-06-15:
> Oops. I just checked the math on https://github.com/
> Smoothieware/Smoothieware/blob/edge/src/modules/robot/arm_solutions/
> RotatableCartesianSolution.cpp and it only rotates. It doesn't correct
> for non-90-degree angles.

Yes, it rotates ( which is what you want in some cases ), but it's be
absolutely trivial to modify to adjust for other corrections.
We don't implement the different cases or document it because it's mostly a
bad idea because of the unintended consequences it would have.

> Such as all machines have to have an exact 90 degree angle mechanically
> built X / Y as software cannot possibly compensate? Have I got it now?
> Thanks Ron
>
> Pretty much. Software compensating for that can create real-world issues.

mayhem2408, 2017-06-15:
@OzRon Let me see if I can explain the problem that lasers would see, with a picture. There is a limitation to the number of steps a stepper/driver can do. If your gantry was out of square and found it was 2mm out at one end, that may give you once 150-200 steps to try and correct it. At each step, you would see an artifact there the previous pixel would not line up. It would show banding and look horrible.

[image]

mayhem2408, 2017-06-15:
Heres a better image
[image]

tbfleming, 2017-06-15:
@mayhem2408 excellent explanation!

mayhem2408, 2017-06-15:
For PnP machines, this works perfectly fine because parts placement will never be more that 1 microstep out, but it just doesn't work with lasering and the same applies to 3D printer. Each microstep compensation will cause a step to appear in the print. This is an exaggerated example of what would happen on a 3D print.
[image]
 I'm not saying that software compensation doesn't have its merits, but it's not for every situation.

OzRon, 2017-06-16:
Thanks everyone for explaining why this is in the too hard basket.  No point in solving one problem only to create more problems.

So I see a perfect machine couldn’t plot a line at say 89 degrees without distortion.  And this is due to resolution?

Pick & Place only needs to move to a single point so that is why non squareness compensation works for that.  And here is me thinking a straight line is just a series of points.  
Now I understand the complexity of plotting a curve from a non-square reference.  It requires far more math and introduces further possibility for error needing very robust coding.

So, having explored the software solution, looks like a couple of days on the mill & lathe to remake gantry ends.  This time using a precision ground toolmakers vice rather than a standard mill vice that ever so slightly cocks the work piece when clamped.  This laser machine is designed to cut over a metre by 300mm for aircraft parts as well as plot precision PCB artwork so I have to get it spot on.

Thanks again to everyone for taking the time to explore the concept and explain the challenges.

Cheers Ron

closed 2017-06-18 by cprezzi