LaserWeb/LaserWeb4 #343: Mid Year recap!

jorgerobles, 2017-06-11:
Hello all @LaserWeb/devteam !

I think is time for a MidYear Recap of Laserweb4 status.
I wish We could put what to be done in next efforts and maybe priorities. No pressure :D

Any issue should narrow use cases if possible (to check if are proven necessary or not like postprocessing)

jorgerobles, 2017-06-11:
For example:

- Integration with more machines and protocols
- Enhance performance of display
- Allow users to customize their workspace
- Some killer functionalities ( Vector packing again ;D )
- Postprocessing.

openhardwarecoza, 2017-06-12:
Fix missing features not yet ported from LW3 (caatch up before we speed onward)

-  Missing Probe / per axis homing comms:  https://github.com/LaserWeb/LaserWeb4/issues/255
- Trace to Vector (popular user request) (poTrace.js)

cprezzi, 2017-06-14:
Probe and per axis homin is in work. ;)

openhardwarecoza, 2017-06-14:
Yay! Oh how i missed probing (;

Sidenote at some point we need to add Probe Diameter for XY probes (actual
probe is at probepos - (probedia /2) since your probe is either an endmill
or a ball probe, both of which has a diameter offsetting actual 0. Maybe
even extend probe with a second menu 'probe and zero' that
1. Probes (stops at end of probe)
2. Sets Zero (but with half probe dia taken into account)

jeepshop, 2017-06-14:
Cool feature. Probably shouldn't use recommend using an end mill as they
actually aren't round. They have a 'wave' to them when looked at from the
side. Below is a common 4 flute end-mill which shows the problem... But
even worse are the 0-flute end mills used to cutting plastics, they aren't
even flat on the bottom. A piece of drill rod will be much more accurate
than an end-mill.

[image: Inline image 1][image: Inline image 2]

tbfleming, 2017-06-14:
Users which have reversing variable-speed spindles could run the mill in reverse at a low speed.

cprezzi, 2017-06-14:
We could add 'probe diameter' in settings, and if the value is 0, use a popup to ask the user for the probe diameter after each button click. Would that be ok?

openhardwarecoza, 2017-06-14:
Yip would be fine.

In terms of hardware precision etc as above, a) irrelevant to the software
side of things. You make your probe run true, we'll send the commands. And
b) remember generic approaches. Some people use endmill, others use
dedicated probes. Some uses completely different things like probes mounted
on the carriages. Irrelevant for our purpose, but it does need an offset
from actual 0 to 'probed' trigger (;

jorgerobles, 2018-04-22:
@tbfleming @cprezzi  Sent you a message by Gplus some days ago. Have you seen it?

tbfleming, 2018-04-22:
I don't see it. Try adding a reply tagging us.

jorgerobles, 2018-04-22:
Done!