LaserWeb/LaserWeb4 #346: svgs importing incorrect size

grennak, 2017-06-15:
i use illustratorcs6 and import to laserweb 4 with svg files, i updated yesterday and now the files import but the size is incorrect, its set to 72 dpi any idea why??

jorgerobles, 2017-06-15:
Could be related to new svg patching @tbfleming ?

tbfleming, 2017-06-15:
Please post an SVG file which does that.

grennak, 2017-06-15:
[silver then gold1406.zip](https://github.com/LaserWeb/LaserWeb4/files/1077548/silver.then.gold1406.zip)
here is a file

tbfleming, 2017-06-15:
Ick. The attributes look like this:

```
width="850.39px" height="566.93px" viewBox="0 0 850.39 566.93"
```

This is what 72 px-per-inch should look like:

```
width="11.81in" height="7.87in" viewBox="0 0 850.39 566.93"
```

I'll have to add a special rule to detect this brain damage Adobe produces and fall back to the configuration setting.

jorgerobles, 2017-06-15:
Sort of. that px are because file is preferenced to use px. I use mm and looks ok. Anyway maybe is more reliable to use an attribute detection to use 72, or disable autodetection and use settings.

jorgerobles, 2017-06-15:
I will test a couple of files later :)

tbfleming, 2017-06-15:
The problem is Inkscape 0.92 has a setting which can produce a header like that. It uses it to imply 96 px-per-inch, which is what all browsers currently assume when width and height units are specified as px. Inkscape versions prior to that always used physical units in the width and height fields. Adobe, when compared to what browsers expect, got it wrong.

cprezzi, 2017-06-15:
How could the user expect a specific size if the file is set to use px? There should not be any inch or mm measurements in Illustrator if set to use px.

tbfleming, 2017-06-15:
Here's where I use the browser's built-in SVG support to convert width to inches:

https://github.com/LaserWeb/LaserWeb4/blob/6c6d59c5a4eb5b636b9ac019f3d0c8447688d369/src/reducers/document.js#L73

> How could the user expect a specific size if the file is set to use px? There should not be any inch or mm measurements in Illustrator if set to use px.

I agree.

cprezzi, 2017-06-15:
This is what actual AI CC creates for a 100x100mm doc with centered 50x50mm square:
```
<svg id="b3b096e4-4a64-4785-ad53-d7591244bc52" data-name="Ebene 1" xmlns="http://www.w3.org/2000/svg" viewBox="0 0 283.46 283.46">
  <title>AI-Test_100x100mm_doc_with_50x50mm_square</title>
  <rect x="70.87" y="70.87" width="141.73" height="141.73" fill="#fff" stroke="#1d1d1b" stroke-miterlimit="10"/>
</svg>
```

tbfleming, 2017-06-15:
@cprezzi it left out the svg tag's width and height fields. That's nice since LW will fall back to the setting when that happens. I wonder if Adobe realized their mistake and fixed it.

grennak, 2017-06-15:
i usually has illustrator set to mm not px but i was trying everything i could and event inches didnt seem to work,  i struggle reading lots of txt. ill let you guys figure this out also i will do more work my end to see what i can figure out. all i did for my files to cut this morning was make them scale in laserweb but that does not help if someone uses or needs exact measurements. ill try checking this every hour or so, please let me know if it gets fixed

tbfleming, 2017-06-15:
@grennak please post one set to mm.

jorgerobles, 2017-06-15:
My fault. I'm reviewing a bunch of files and all are in px :|
I mistook the file with an Inkscape one

grennak, 2017-06-15:
[silver then gold1406.zip](https://github.com/LaserWeb/LaserWeb4/files/1077839/silver.then.gold1406.zip)

cprezzi, 2017-06-15:
@tbfleming By the way: my code was created with the flag "use workspace" activated.

tbfleming, 2017-06-15:
@cprezzi I never heard of that option. What does it mean?

tbfleming, 2017-06-15:
Dang it. @grennak 's mm file still says this:

```
width="850.39px" height="566.93px" viewBox="0 0 850.39 566.93"
```

tbfleming, 2017-06-15:
I'm thinking of creating this checkbox in settings:

```
Adobe Crap
```

This will bypass px/inch detection.

cprezzi, 2017-06-15:
@tbfleming The feature includes the whole document (size) instead of just the objects in it.

cprezzi, 2017-06-15:
We aware that Illustrator CS6 is very old!

jorgerobles, 2017-06-15:
I've got this (fragment)

```
<?xml version="1.0" encoding="utf-8"?>
<!-- Generator: Adobe Illustrator 19.0.0, SVG Export Plug-In . SVG Version: 6.00 Build 0)  -->
<svg version="1.1" id="Capa_1" xmlns="http://www.w3.org/2000/svg" xmlns:xlink="http://www.w3.org/1999/xlink" x="0px" y="0px"
	 width="595.28px" height="841.89px" viewBox="0 0 595.28 841.89" enable-background="new 0 0 595.28 841.89" xml:space="preserve">
<polygon id="XMLID_27_" fill="#FFFFFF" stroke="#000000" stroke-miterlimit="10" points="119.591,100.748 206.699,13.641 
	293.806,100.748 "/>

```

jorgerobles, 2017-06-15:
Do It !

tbfleming, 2017-06-15:
@jorgerobles your fragment also has the issue :(

jorgerobles, 2017-06-15:
Yes. The best will be or setting autodetection or dropping it :/

tbfleming, 2017-06-15:
The setting is "Force PX Per Inch". We can document it as the flag to support old AI versions.

tbfleming, 2017-06-15:
New recommendations:
* Inkscape users, both new and old versions:
  * Don't tick the box
  * PX Per Inch setting doesn't matter
* Old AI versions
  * Tick the box
  * PX Per Inch = 72
* New AI versions
  * Don't tick the box
  * PX Per Inch = 72

tbfleming, 2017-06-15:
Short term it wouldn't hurt if all AI users ticked the box. Long term that recommendation will backfire if AI modernizes its conversion.

grennak, 2017-06-21:
i have done some testing on my end , did the update stop using the setting for import dpi, as regardless of what it says it imports at 96dpi, you wouldnt need a setting for each software if you took the import dpi from that setting again

tbfleming, 2017-06-21:
I don't understand what you just wrote

grennak, 2017-06-21:
in the settings for lw4 there is a setting for import dpi of svg files, if the update stopped using this and instead tries to get it from the svg header why not just have it use that dpi again, i have tried changing it and it does not effect it, so what ever was changed must have bypassed this setting

tbfleming, 2017-06-21:
Did you tick the box? That tells it to ignore that bad info AI puts in the header.

jorgerobles, 2017-07-10:
@grennak does this still an issue? Please test latest binary and close if fixed. Thanks

grennak, 2017-07-10:
everything seems fine now thank you

closed 2017-07-10 by jorgerobles