LaserWeb/LaserWeb4 #347: Recent changes tab for LW app

cojarbi, 2017-06-15:
I propose a new tab at the end after About to address user with latest changes in the app including:

New features
New config values
Bug fixes
Etc

This could reflect directly from commits comments ( just need to be expanded a little bit ) and could be an intermediate step to getting documentation into cncpro web. 

This would help propagating changes that need user input like the adobe crap in here #346

openhardwarecoza, 2017-06-27:
API: https://api.github.com/repos/LaserWeb/LaserWeb4/commits?sha=dev-es6

jorgerobles, 2017-06-28:
I would look for the api call to get the release comments rather than
commits

openhardwarecoza, 2017-07-24:
Parse https://api.github.com/repos/LaserWeb/LaserWeb4-Binaries/releases to get the releases. 

Each child in the object is a release. 

Each release has a release tag_name

for example: 
```
"tag_name": "v4.0.989-115",
"tag_name": "v4.0.988-114",
"tag_name": "v4.0.736-114",
```

and "body"

```
    "body": "Frontend:\r\n[FIX] - Material database picker error.\r\n[ENH] - Better DXF handling \r\n[FIX] - Wrong Raster OP first movement \r\n\r\nBackend:\r\n[ENH] Improved feature detection\r\n[FIX] Work offset logging refactor."

"body": "Bug fix release:\r\n\r\n* Some SVG files had issues\r\n* CAM webworkers failed to initialize\r\n"
  },

   "body": "````\r\n---\r\nWE ARE WORKING TO FIX SOME ISSUES ON THIS RELEASE. STAY TUNED\r\n---\r\n````\r\n\r\nFrontend:\r\n - [Upg] Upgraded SnapSVG, improves svg import.\r\n - [Enh] Improved DXF loading\r\n - [Enh] detachable/persistent Floating controls.\r\n - [Enh] Material database deleteable examples\r\n - [Fix] Material database overwriting user profiles on certain conditions.\r\n - [Fix] Material database not honoring operation type on applying presets.\r\n - [Fix] Minor UI improvements (command-history, floating windows, help balloons...)\r\n - [Fix] Simtime with display_cache=on\r\n\r\nBackend: \r\n - [Enh] New command setPosition for upcoming changes.\r\n - [Fix] Error log generation\r\n - [Fix] Smoothie Firmware detection"
```

jorgerobles, 2017-07-24:
Great!!

openhardwarecoza, 2017-07-24:
[image]

Assets contain the download links. Might even be able to use that to semi-automate updates?  (A new version is available, do you want to update? -> electron downloads and exec() the package?

jorgerobles, 2017-10-09:
Done!

closed 2017-10-09 by jorgerobles